x-station/xstation-issues #123: Booting Memory Card Screen

trev1976, 2021-05-31:
Hi 

I installed my xstation today and followed the guide but I'm just getting the memory card screen.

Triple checked my solder points with a multi metre and all seems good , tried 3 different sd cards but still nothing. 

I get the solid blue led and then a solid green led before the memory card screen.

I have the door closed too

Any ideas 

Thanks

trev1976, 2021-05-31:
Do you have to have games on the SD card for it to boot ? 

Thanks

ramapcsx2, 2021-05-31:
Hi,
you have to have at least the 00xstation folder on your sd card, with the loader.bin file in it.
This is enough to get it to boot into the loader.

For tips and tricks, please have a look at all the other (closed) issues with the same description.

trev1976, 2021-05-31:
Ok will do , I've got the 00xstation on the sd card plus loader in the folder.

ramapcsx2, 2021-05-31:
Here is a link to display all old issues, just browse through some that fit the error descriptions and you'll find a lot of suggestions there.
Basically, the error could be almost anything at this point :p 
https://github.com/x-station/xstation-issues/issues?q=

trev1976, 2021-05-31:
OK thanks,  yes very annoying problem,  solder points look good with nothing touching each other and multi metre checks out.

Don't fancy taking the qsb off if I don't have too.

trev1976, 2021-06-02:
Hi 

Took a couple of videos to show the led's and the QSB soldered on, Still can't get it to boot, reflowed several times.

https://youtu.be/vxZtAlkJeos

https://youtu.be/zINl1_hGIFo

ramapcsx2, 2021-06-02:
The videos are marked private. I can't watch them.
But in any case, what often happens is that people leave solder bridges between the lifted DSP pins, or otherwise short 2 signals together. Make sure to test for that.

trev1976, 2021-06-02:
OK sorry about that, changed them to unlisted now.

trev1976, 2021-06-02:
Finally got it working,  went for one last reflow on the qsb and boom .. so wired the yhe continuity was checking out but it still wouldn't work. Just enjoying some RE ultimate directors cut on my PVM. :)

closed 2021-06-02 by trev1976