x-station/xstation-issues #127: PU-18 boots to BIOS/CD Player - Was working for weeks.

bootleg714, 2021-06-04:
Hi,

I have a PU-18 that has been working for weeks. Now it only boots to the CD Player. I've installed several of these and Ive never come across this issue. I've quadruple checked all connections, tested the lid switch and read most of the threads.

The only thing that I've found is I'm not getting a signal from pin 20 to the mechacon. I read that could also cause problems in one of the posts. Does anyone know which pin on the mechacon is supposed to connect to pin 20 on the ribbon cable clamp?

Thanks for any help!

ramapcsx2, 2021-06-04:
20 is the limit switch on the drive. It is required to boot, so the missing signal explains the problem.

bootleg714, 2021-06-04:
Do you happen to know which pin on the mechachon pin 20 needs to connect to? Thanks again!

bootleg714, 2021-06-04:
UPDATE:

Its pin 5  from the left on the bottom row of pins on the chip that needs to connect to pin 20. I ran a wire and it booted up first try. 

I hope this help other with this issue. Thanks for Rama for the info!!

ramapcsx2, 2021-06-04:
Glad it wasn't too hard to fix then :)

closed 2021-06-04 by ramapcsx2