x-station/xstation-issues #128: Mega Man X4 PAL not working

msimplay, 2021-06-05:
My PS1 is a Pal Late PU 8 with the latest 1.30 firmware and the game doesn't boot up.
I have tried both fast and full boot and the memory card was formatted with the SD Card Formatter

Redump Mega Man X 4 but the NTSC redump works fine

ramapcsx2, 2021-06-05:
I'll take a look. Stay tuned!

ramapcsx2, 2021-06-05:
Very interesting find!

This game might shed some light on badly documented CD structure details.
I can fix this, but I'll have to make sure the fix doesn't break any other titles.

It'll work with the next update :)

ramapcsx2, 2021-06-05:
Hmm, the game will not run at all.
I think the redump might be bad at this point.
It crashes on real hardware when played from disk, too.

msimplay, 2021-06-05:
I'll make a dump from my original disc and see if it works.

I'll be testing all of my pal games and keep posting here