x-station/xstation-issues #131: Tomb raider 2 - 5 runs too fast?

BrainFreeze311, 2021-06-12:
Played through the original tomb raider fine, no speed issues.
Tried TR2-5 and they all seem to run about 10% too fast. Not sure what the problem is.

PAL PU-18 motherboard, latest firmware. NTSC games.

ramapcsx2, 2021-06-12:
I haven't noticed anything like this and there should be no reason for it either :)

BrainFreeze311, 2021-06-12:
Ok, ill try another rom, maybe i have a bad rip for these.

ramapcsx2, 2021-06-17:
Is Lara back to normal speed now? :)
It's a weird bug that can't be caused by the ODE at least.

BrainFreeze311, 2021-06-27:
Nope, still the same issue, I was hoping that someone else might of had a similar problem, It’s weird as the cutscenes seem to run at normal speed, but the gameplay is running too fast.
Ive tried different ROMs, even tried ripping my PAL copy of TR2, still the same issue.

It’s not the ODE, at least, as every other game seems to run fine.

Thanks anyway.
- Tom

retrofank, 2021-06-28:
NTSC will run faster than it's PAL counterpart.  However, also running a NTSC title on a PAL unit will run slightly out of spec, so ideally a dual DFO is needed to have a true NTSC signal.