x-station/xstation-issues #132: Elemental Gearbolt won't launch

les-chenis, 2021-06-16:
Hi team,
I can't launch Elemental Gearbolt (NTSC-U) on my PAL 5502 PU-18 model, that's the only NTSC game I can't boot.
I tried at leat 5 different bin/cue but each time I have a black screen after the Playstation logo

Any idea ? Some region-protection maybe ?

Except this one, every game I tested (approx 100 different, each in PAL and NTSC) works well, such a great product !

ramapcsx2, 2021-06-16:
This is one of the 2 games that don't work with fast boot currently. Please use full boot for it :)
https://github.com/x-station/xstation-issues/issues/8

les-chenis, 2021-06-16:
You rule, it works great with full load :)
Sorry for the irrelevant issue report...

closed 2021-06-16 by les-chenis