x-station/xstation-issues #133: PAL Games over 200 discs compatibility list tested Final Fantasy 8 (SLES-02080) not working

msimplay, 2021-06-17:
Games of Note
[]Final Fantasy 8 (not working redump, cue/bin or ccd black screen only) (SLES-02080)
[x]Capcom vs snk pro - working fine but for some reason IGR didn't work 😕
[x]Mega Man X 4 (redump not working had to rip my own copy to ccd format not able to rip to redump format someone else could check?)
[x]Point Blank 2 black screen after selecting language used full boot after normal boot failed redump copy didn't work had to dump own to ccd

All Working no problems PAL Games on a Late PU-8 1002 PS1 with IGR

[x]Air Combat - Requires full boot
[x]Ace Combat 2 - works fine
[x]Alundra - works fine
[x]Alundra 2 - works fine
[x]Ape Escape - works fine
[x]Arcade Party Pak - tested all games on the pack works fine
[x]Battle Arena Toshinden 
[x]Blazing Dragons
[x]Bloody Roar 2
[x]Breath of Fire 3
[x]Bushido Blade
[x]Bust a move 2
[x]Bust a Move 3DX
[x]Bust a Move 4
[x]Capcom Generations tested all games on all discs working 
[x]Castlevania Symphony of the Night 
[x]Castlevania Chronicles - working both arrange and original 
[x]Chrono Cross disc 1 (USA)
[x]Chrono Cross disc 2 (USA)
[x]Civilization II
[x]Colin McRae Rally 1.0 
[x]Colin McRae Rally 2 1.0
[x]Command and Conquer
[x]Command and Conquer Red Alert
[x]Crash Bandicoot
[x]Crash Bandicoot 2
[x]Crash Bandicoot 3 Warped
[x]Croc Legend of the Gobbos
[x]Croc 2
[x]CTR Crash Team Racing
[x]Cyberbots
[x]Darkstalkers
[x]Darkstalkers 3
[x]Dead or Alive
[x]Destruction Derby
[x]Destruction Derby 2
[x]Destruction Derby raw
[x]Discworld
[x]Discworld 2
[x]Driver
[x]Exhumed
[x]Fifa Road to the world cup 98
[x]Final Fantasy 7
[x]Final Fantasy 9 (requires full boot)
[x]Final Fantasy Tactics (USA)
[x]Formula 1
[x]Formula 1 97
[x]Formula 1 2000
[x]Front Mission 3
[x]G-Police both discs
[x]G.Darius music seems messed up but it's actually the same on the original disc also
[x]Gekido
[x]Gran Turismo 2
[x]Grand Turismo EDC 
[x]Grandia disc 1 and 2
[x]Guilty Gear
[x]Heart of Darkness disc 1 EDC
[x]Heart of darkness disc 2 
[x]Hogs of War
[x]In the hunt
[x]International Track and Field
[x]Iron Soldier 3
[x]ISS Pro Evolution 2
[x]Jumping Flash
[x]Jumping Flash 2
[x]King of Fighters 95
[x]King of Fighters 96 (JAP)
[x]King of Fighters 97 (JAP)
[x]King of Fighters 98 (JAP)
[x]King of fighters 99 (USA)
[x]Legacy of Kain Soul Reaver
[x]Lemmings and oh no more lemmings
[x]Lomax
[x]Magical Drop 3 disc 1
[x]Magical Drop 3 disc 2
[x]Marvel Super Heroes
[x]Marvel Super heroes vs Street Fighter
[x]Marvel Vs Capcom
[x]MediEvil
[x]Medievil 2
[x]Metal Gear Solid disc 1 and 2
[x]Metal Slug
[x]Metal Slug X
[x]Mickeys Wild Adventure
[x]Micro Machines v3
[x]Monaco Grand Prix Simulation 2
[x]Mortal Kombat 4
[x]Mortal Kombat Trilogy
[x]Moto Racer
[x]Moto Racer 2
[x]Motor Toon Grand Prix 2
[x]NBA Jam T.E sound cut out try original game
[x]Omega Boost
[x]Pandemonium
[x]Parodius + parodius fantastic journey
[x]Pocket Fighter
[x]Primal Rage
[x]Psychic Force
[x]Psychic Force 2
[x]Project Horned Owl (US)
[x]R-type Delta
[x]R-Types - R-Type 1 + 2 work
[x]Rage Racer
[x]Raiden Project ( needed a full load but both games working )
[x]Ray Tracers
[x]Rayman
[x]Rayman 2
[x]Raystorm
[x]Ready 2 Rumble Boxing
[x]Real Bout Fatal Fury
[x]Realbout Special Dominated Mind (JAP)
[x]Resident Evil EDC
[x]Resident Evil 2 disc 1
[x]Resident Evil 2 disc 2
[x]Resident Evil 3
[x]Ridge Racer
[x]Ridge Racer Revolution EDC
[x]Ridge Racer Type 4
[x]Ridge Racer hi-spec demo
[x]Rival Schools Evolution disc
[x]Rival Schools disc 2
[x]Samurai Shodown 1+2 - Kenkaku Shinan Pack (JAP)
[x]Samurai Shodown 3
[x]Samurai Shodown IV Amakusa revenge special (JAP)
[x]Silent Bomber
[x]Silent Hill
[x]Silouhette Mirage
[x]Soccer Kid
[x]Sonic Wings Special
[x]Soul Blade
[x]Speed Freaks
[x]Spyro 2 Gateway to glimmer
[x]Spyro the Dragon
[x]Spyro Year of the Dragon
[x]Street Fighter Alpha
[x]Street Fighter Alpha 2
[x]Street Fighter Alpha 3
[x]Street Fighter Collection disc 1
[x]Street Fighter Collection disc 2
[x]Street Fighter Collection 2
[x]Street Fighter EX Plus Alpha
[x]Street Fighter EX2 Plus Alpha
[x]Street Racer
[x]Strider 2
[x]Strider 
[x]Strikers 1945 (USA)
[x]Strikers 1945 2
[x]Super Puzzle Fighter II Turbo
[x]Syphon Filter 2 disc 1
[x]Syphon Filter 2 disc 2
[x]Tekken
[x]Tekken 2
[x]Tekken 3
[x]The Adventures of Alundra
[x]The Need for speed
[x]Theme Hospital
[x]Theme Park
[x]Time Crisis
[x]Time Crisis Project Titan
[x]Toca 2 Touring Cars 
[x]Toca World Touring Cars
[x]Tomb Raider (EDC) 
[x]Tomb Raider 2
[x]Tomb Raider 3
[x]Toy Story 2
[x]Tony Hawks Pro Skater 2
[x]Vagrant Story
[x]V-Rally 2
[x]Wipeout 1.0
[x]Wipeout 2097
[x]Wipeout 3 Special Edition
[x]Wipout 3
[x]Worms
[x]Xenogears (USA)
[x]Xmen Cota
[x]Xmen vs Street Fighter

ramapcsx2, 2021-06-17:
Oh, this is a nice list!
I always forget adding FF8 regional variants to new firmwares. 
The problem is missing libcrypt entries that have to be added, and I'll get to it :)

msimplay, 2021-06-17:
Yes I thought most people will be going for NTSC games but I just wanted to play my PAL games on sd card so maybe this will be useful for people.
Especially since there are a few PAL exclusives

oldmodder, 2021-09-28:
Hello everyone, any news on a new release have the same issue with the FF8 title,

Also will it be possible or it is too much work / complication to implement an external file to source for those title we can add and remove them in case of missing one for example :)

Thanks for your work @ramapcsx2

msimplay, 2021-09-28:
I can't remember which one but I used one of these patches to fix final fantasy 8

https://consolecopyworld.com/psx/psx_patches_f.shtml

oldmodder, 2021-09-28:
@msimplay yeah that the old school way (and the old school site :P) and a good workaround, thanks for the input, but was asking for native support (dynamic patching ), again thanks for the advice :)

msimplay, 2021-09-28:
Yeah I think he said he normally adds these patches to the firmware normally they are in for the US version but missing on PAL for now at least so likely in the next update

missing libcrypt entries he said 
https://github.com/x-station/xstation-issues/issues/133#issuecomment-863129967

oldmodder, 2021-09-28:
Yeah seen the post :) been shadow following the thread since jun, just wanted to ask kindly of a status update and maybe share an idea/change to externalize that so it's not dependent on the firmware binary.

retrofank, 2021-09-28:
Join the Classic Gaming discord and download the Xstation beta updates.  FF8 PAL LC is now fixed in beta.  Imo, I wouldn't be using cracked scene release patches now, since ideally you want to be using clean images.

closed 2021-12-03 by ramapcsx2