x-station/xstation-issues #135: cheesy PAL Game

pcaudill3323, 2021-06-19:
the PAL Game Chessy does not work at all?

ramapcsx2, 2021-06-19:
No problems with it here. Is your dump of the game correct?

pcaudill3323, 2021-06-19:
Yes. But I do have the mod chip in stalled to play pal games in color but it did not play it before the mod chip.

From: Robert Neumann
Sent: Saturday, June 19, 2021 11:43 AM
To: x-station/xstation-issues
Cc: pcaudill3323; Author
Subject: Re: [x-station/xstation-issues] cheesy PAL Game (#135)

No problems with it here. Is your dump of the game correct?

ramapcsx2, 2021-06-19:
Do other games work fine?
I don't know about that other chip and how it would affect things. Please explain more about it.

pcaudill3323, 2021-06-19:
All the other games work and the chip allows PAL game to play in NTSC I think it’s a MM3 mod chip.

From: Robert Neumann
Sent: Saturday, June 19, 2021 3:27 PM
To: x-station/xstation-issues
Cc: pcaudill3323; Author
Subject: Re: [x-station/xstation-issues] cheesy PAL Game (#135)

Do other games work fine?
I don't know about that other chip and how it would affect things. Please explain more about it.

ramapcsx2, 2021-06-19:
I'm speculating now, but if you had the xStation installed by a modder, he probably disconnected the actual modchip function and only left the color clock modification.
A modchip would interfere with the xStation's operation, and would not even allow the quick solder board to be installed correctly.

So .. I don't know what to do to help.
The game is working fine for me. There is nothing special about it.

pcaudill3323, 2021-06-19:
I downloaded the japan version, it works, I guess the (EU) version is encoded.

From: Robert Neumann
Sent: Saturday, June 19, 2021 4:39 PM
To: x-station/xstation-issues
Cc: pcaudill3323; Author
Subject: Re: [x-station/xstation-issues] cheesy PAL Game (#135)

I'm speculating now, but if you had the xStation installed by a modder, he probably disconnected the actual modchip function and only left the color clock modification.
A modchip would interfere with the xStation's operation, and would not even allow the quick solder board to be installed correctly.
So .. I don't know what to do to help.
The game is working fine for me. There is nothing special about it.

ramapcsx2, 2021-06-19:
Mine is the PAL version. There's nothing special I can see.

ramapcsx2, 2021-06-25:
If you have an expansion ROM cart attached, try without it :)

ramapcsx2, 2021-08-05:
I'll close this, unless the issue reappears somehow.
I found that on my machine, the game wouldn't work with an expansion cart inserted.

closed 2021-08-05 by ramapcsx2