x-station/xstation-issues #136: Massive cheating issues (xstation + ps1digital)

RealRaptoReX, 2021-07-02:
I have a serious problem with using cheats on my hardware. Playing games works normally, but cheating doesn't.

A week ago, my xploder cd 9000 worked normally. I had the issue of Gran Turismo 2 (PAL) not reacting to any codes, so i ordered a xploder fx cartridge for the parallel port. After many cheating attempts, both the cd (bin/cue) and the cart don't work anymore. The card gets a black screen and the cd hangs at the loading game screen now.

Before this happened, if i activated cheats, they would work, but with the cart, the games ran slower, were prone to crashing completely and worst of all, couldn't access any memory cards. I also had very massive slowdown and graphical artifacts on screen.

I don't know why this happens and i tried different region roma and cheat codes, but mow i can't cheat at all. Can anyone explain or think of a reason why this hapens? Any help would be apreciated, because i'm going insane with this.

ramapcsx2, 2021-07-02:
Yea, I recognize this:
Your ROM cart (Xploder?) has a mode active that allows easier cheat searches in games.
This adds extra code to the game that slows everything down and breaks many titles completely.
In your cart menu, disable the button combination (set it to "none" or "N/A") for entering the in-game menu.

RealRaptoReX, 2021-07-02:
> Yea, I recognize this:
> Your ROM cart (Xploder?) has a mode active that allows easier cheat searches in games.
> This adds extra code to the game that slows everything down and breaks many titles [...]

Yeah its an Xploder FX cheat cartridge for the parallel port. But i never saw a button combination oprion for in-game menu access. Not on this and not on the cd 9000. Besides that, now i have the issue of the cart only booting to a black screen after i put it in and turn it on.

ramapcsx2, 2021-07-02:
The connector doesn't make good contact then.

RealRaptoReX, 2021-07-02:
> The connector doesn't make good contact then.

I don't think so. Even CD based cheat systems crash, none of them work anymore all of a sudden.

ramapcsx2, 2021-07-02:
Well, one thing is for sure: It's not an xStation bug :)

RealRaptoReX, 2021-07-02:
> xStation

Any ideas what else it could be? It happened after many attempts. All of a sudden no cheat cd or device works with black screens and crashes. Maybe something with the memory? I can't think of any reason this hapened.

ramapcsx2, 2021-07-02:
That's hard to say, especially since I don't know whether you have your Xploder FX plugged in or not.
If it's plugged in, the feature I explained will definitely cause this.
If it's not plugged in, and everything works EXCEPT cheating from the disc based one, then that's unexplainable.

RealRaptoReX, 2021-07-02:
> That's hard to say, especially since I don't know whether you have your Xploder FX plugged in or not.
> If it's plugged in, the feature I explained will definitely cause this.
> If it's not plugged in, and everything works EXCEPT cheating from the disc based one, then that's unexplainable.

If it is plugged in, leaving it off doesn't do anything, when its on, its just a black screen. I didn't toggle that feature you talked about so that's also confusing me. The cheat discs don't work anymore no matter if that thing is plugged in or not.

ramapcsx2, 2021-07-03:
Until you have solved this, leave the Xploder FX unplugged. Otherwise, the software on it will always try to install the code hooks and crash your software.

Next, you may be doing something wrong (different) in the CD based version. I really can't tell though. It's not something that the ODE could cause or affect.

RealRaptoReX, 2021-07-03:
> Until you have solved this, leave the Xploder FX unplugged. Otherwise, the software on it will always try to install the code hooks and crash your software.
>
> Next, you may be doing something wrong [...]

Maybe it did install some faulty code hooks?
It happens with CDs even when i select no codes and straight up start the disc.
Could it have something to do with the ps1digital and its settings?

nicolasnoble, 2021-07-03:
> Maybe it did install some faulty code hooks?
There is no code hooks for the x-station.

RealRaptoReX, 2021-07-03:
> > Maybe it did install some faulty code hooks?
> 
> There is no code hooks for the x-station.

I remembered something. I did use that function in the reset menu of the ps1digital, something like realign memory something. Maybe it was that? At this point i'm speculating, can't think of anything else.

ramapcsx2, 2021-07-03:
That ps1digital option might have been for the GPU clock. It might "align" to PAL/NTSC timings.

RealRaptoReX, 2021-07-03:
> That ps1digital option might have been for the GPU clock. It might "align" to PAL/NTSC timings.

So it isn't that either. I'm out of ideas. Isn't there a way to reset the xstation to factory settings? Maybe that would help?

ramapcsx2, 2021-07-03:
It cannot be the xStation :)
But if you want to try other firmware revisions, you may install any released update you want.
Be aware that you should use the loader.bin that came with each release.

RealRaptoReX, 2021-07-03:
> It cannot be the xStation :)
> But if you want to try other firmware revisions, you may install any released update you want.
> Be aware that you should use the loader.bin that came with each release.

I was just being stupid, i forgot to boot the cheat system and the game as multidisc. The cd based ones, at least the action replay 4.2 uk version, works as intended.

But the xploder fx still doesn't work, probably busted. Still black screen for some reason.

ramapcsx2, 2021-07-03:
If the FX doesn't even start up, one or more pins don't make contact. It's very common with the age of the machines and carts today.
Repeated (often) inserts and removals can fix it.

ramapcsx2, 2021-07-04:
Also a bit of good old electrical contact cleaner (oxide dissolver) does wonders :)

I'll close this, as it's not xStation related :p

closed 2021-07-04 by ramapcsx2