x-station/xstation-issues #137: Quake II intro bug

mario-64, 2021-07-06:
There is a bug with the Quake II intro which causes the video to be split in half top to bottom with a large black bar in between. Tested on firmware 1.2.0 and 1.3.0. Confirmed it is coming from Xstation by testing the same image on PSIO. No issues there. Actual gameplay seems ok. Bug is with the intro. See attached photo. Please forgive my terrible photography skills.

[image]

ramapcsx2, 2021-07-06:
Hey,
as discussed on the Discord, it seems to be fine for me and others.
Might be ps1digital related, though I couldn't say why only on xStation.

(Maybe there is an automatic patch in ps1digital to set something video related, which PSIO doesn't support yet.)

mario-64, 2021-07-06:
Could be I suppose. Confirmed that setting OSD->Capture Adjust Settings->Vertical to 0 fixes it.  Strange that the issue only manifests when using Xstation though. Either way, all good now. Thanks for the help.

closed 2021-07-06 by mario-64

ramapcsx2, 2021-07-06:
Sure.
I bet it has to do with the game ID feature :)