x-station/xstation-issues #138: Random Resets

Dethslayer85, 2021-07-21:
Hi, I just installed my 2nd X-Station into a PU-18 and games are randomly resetting. I searched online to no avail. I re-soldered all of the contact pins on the board and it is still happening. It 1st happened on Clock Tower (US) then Chrono Cross (US) both during the opening FMV. Since then it has happened on MGS (US) and Valkyrie Profile (US) during gameplay. It doesn't happen often but it is becoming tiresome. Any help would be greatly appreciated!

ramapcsx2, 2021-07-21:
Hey, if you have another power supply, please try swapping them.
Randomly resetting might be related to capacitors, on the power supply and/or the mainboard.
It might also be a worn and oxidized power supply reset button. This often improves when pressing the reset button several times while the console is off.

Dethslayer85, 2021-07-21:
Great advice! I will try and get back to you! I happen to have 3-4 5501's laying around.

Dethslayer85, 2021-07-22:
I tried giving the reset button a workout and it hasn't reset randomly since. Tyvm for your help! I will close request! 🙌👌👍🙏

closed 2021-07-22 by Dethslayer85

ramapcsx2, 2021-07-23:
Oxidation strikes again :)