x-station/xstation-issues #139: PU-18 boots to Memory Card / CD Player after trying to transfer saves from mem card 2 to mem card 1

hakkachu, 2021-07-22:
Currently on latest firmware 1.3 running on 5501 PU-18. I'm using Memcard Pro (On latest firmware) in slot one and OEM PSX memory card in slot 2.

Xstation worked up until today when I decided to leave the PSX CD lid open to access the memory card menu to transfer save files from Mem card 2 to Mem Card 1.  I then powered off and on resetting the PSX and it would no longer boot into Xstation and I was stranded at the Mem Card / CD player screen each time as if the CD lid was still not closed. I looked on here for solutions and follow the troubleshooting steps as some of us here went through with the same experience where Xstation has solid blue light and the green blinks trying to access the SD card with the CD lid button held down when powered on. Thinking I may be a victim of the dreaded SD slot malfunction problem, I decided to reflow all of my QSB solder points first but that didn't resolve the issue. I then performed continuity test at all of the points using the available troubleshooting diagram, flex cables, micro sd card slot and yes everything beeped so connections should be all good. I then checked for bridged contacts on the QSB and found 14 (RED) and the GROUND (YELLOW) next to it had continuity _**(I circled in green below)**_. Sucked the solder off both points and re-soldered and no more continuity. Xstation 1.3 boots up and plays just fine again. Just out of curiosity I tried to replicate the issue again by transferring my save from Mem card 2 to Mem card 1 and once again could not boot back into Xstation. Opened up my PSX and once again found solder points 14 (RED) and GROUND (YELLOW) next to it was beeping when I checked for continuity, Sucked the solder at those 2 points again and reflow new solder and no more beeping. Xstation 1.3 boots up again and play just fine. This is definitely repeatable as I did this at least 3 times as I got even better at taking apart my 5501 and got even more familiar with my PU-18 :)

I'm just a hobbyist enjoying the latest tech on retro consoles and it has been amazing. But have I stumbled upon a hardware fault or bug tripping a fuse on the PU-18 somewhere at solder point 14? Or is this normal?

[image]

ramapcsx2, 2021-07-22:
Hey,
great that you were able to find the problem! This would have been difficult to debug.

The problem is entirely with the QSB connecting point 14 to the other point after a while.
I assume the QSB underside or the PSX board developed a tiny hair crack, possibly with a bit of solder in it.
When you reflow the area, it clears up, but this is only temporary. After waiting a while or possibly after it warms up from use, the 2 points connect again.
It's coincidence that it happens while you transfer saves :)

I would recommend you carefully unsolder all the lower points, so you can lift the point 14 area a little and clean up whatever is in there.
Once this is done, the install should work properly.

Best of luck!

hakkachu, 2021-07-23:
Hello there!

I really appreciate your quick response. Your mention of the possible root causes sound very valid and plausible indeed. I will need to do a bit of work to investigate and clean up underneath that point 14 area. I might end up attempting to remove the whole QSB just to make sure, which I'm not looking forward to do. It is really soldered on there after my last couple reflows.

Thank you for giving my issue your attention and your invaluable experienced suggestions to the possible root cause. It always helps to understand and know the issue to properly resolve the problems 👍 .

closed 2021-07-23 by hakkachu