x-station/xstation-issues #140: Stuck No Games Found

dillysd, 2021-07-31:
I installed an xstation into a PU-18 board. It keeps getting stuck at the blue screen with No Games Found, Press (^) to refresh list. I press the 

Things I have tested:
I have tried 3 different SD cards
With and without laserbear addon
2 different PU-18 boards
Bought an additional daughter board for PU-18 and soldered that on
Tested all the points at the board, solder joints and the ribbon cable up to the xstation

What am I missing here?

ramapcsx2, 2021-07-31:
The installation seems to be fine, otherwise you wouldn't get into the loader.

The games should be in .bin/.cue format (alternative: .iso or .img).
Make sure the SD card is not write protected by chance.

dillysd, 2021-07-31:
Thanks for responding.
I checked the SDs for read write and they are good.
The games are all in bin/cue. 

I switched out the controller ports and cable as a test as well. Sadly, just now stuck with a dud…

ramapcsx2, 2021-07-31:
It must be some kind of SD card layout problem, otherwise you wouldn't get into the loader.

Your system folder is /00xstation  ?
The games are <outside> this folder?

dillysd, 2021-07-31:
Yes, this is how it’s currently set up on two different cards for testing.

ramapcsx2, 2021-07-31:
Have you updated the firmware yet?

badass-boss, 2021-08-01:
Just fully format (don't quick format) your SD card and write files all over again. This usually the solution for these kinds of problems

dillysd, 2021-08-02:
> Have you updated the firmware yet?

I am using update 130. I was trying to get this to update but I can't get to that screen. I must be losing access to the SDcard

dillysd, 2021-08-02:
w

> Just fully format (don't quick format) your SD card and write files all over again. This is usually the solution for these kinds of problems

I'm changing from my Windows PC SDcard.org formatter to the Mac SDcard.org formatter to see if this changes the game. Hopefully, I want to play Castlevania badly!!!

badass-boss, 2021-08-02:
Windows version should still do it. Maybe the source of your game files is bad. If you downlo.... ekhm... buy games from piratebay or similar websites then there is high chance those files are corrupted. They are often badly ripped, mismatched filenames, mismatched checksum etc. Just rip some of your original game disks (or at least any disks that are 100% confirmed working) to bin + cue format and try that

dillysd, 2021-08-02:
Makes sense. I keep it legit and legal. I created backups (.bin/.cue) from my host copies and tested them on PCSXR. They work great. I have a feeling it might be that windows machine. I

birdybro, 2021-08-02:
Use the cuesheets from redump.org --> http://redump.org/downloads/ - and then rename your .bin files to match what it says in the .cue files from redump.

Cuesheet syntax matters a ton.

ramapcsx2, 2021-08-02:
Hey guys,
these tips are all valid and good, but the user has some other issue.
His loader starts, so the xStation is definitely working and the SD card does at least the minimum work of providing a file system and reading the loader.bin image.
What fails is either the creation of a games.txt (maybe write protection), or the scanner can't find files.

It is important that the firmware is up to date, so it matches the loader.bin!
XStations ship with 1.0.3 firmware, which will not scan the SD card properly when using a recent loader.bin.
The user should press Triangle and go directly to "Update Firmware". This works with any loader.bin/firmware combination.

dillysd, 2021-08-02:
Never thought of this, I’m going to go through the old firmware and see if it loads and creates txt so I can get it updated to the newest. 

Be back soon!!

badass-boss, 2021-08-03:
Did you even read the manual?? The previous user told you to update the firmware, not use an older version. The manual also clearly states to update the firmware with steps how to do that

ramapcsx2, 2021-08-03:
@dillysd 
Please download this: https://github.com/x-station/xstation-releases/releases/download/1.3.0/update130.zip
Extract the 2 files to your micro SD card, into a directory named 00xstation.

Insert that micro SD card into your xStation and boot up. The loader should come up.
Press triangle on your controller to enter the options screen, then navigate to "Update Firmware" and select it.
Press Start on the controller to confirm the update and wait for it to complete and reboot the console.

I've typed this out, just in case there is still a misunderstanding. Please report back how it went :)

dillysd, 2021-08-03:
I went back through all the firmwares since its been about 2 months since I did the original install. I am getting stuck at the loader screen with all levels of firmware. I go to press triangle and nothing happens. I did confirm the txt file is being created on the SD card. Thanks for the response, I did go through the original documentation prior to the install.

ramapcsx2, 2021-08-03:
Okay, it really was unclear whether you did the update or not.

Do any of the other buttons do anything?
At this point I'd like to know whether controller inputs are even recognized.

badass-boss, 2021-08-04:
This is what I thought about too. Is this possible that some PCB trace was broken by accident?? Also, does Xstation require to be connected to controller's serial pins? Overall, what kind of signals is it connected to? Anyway, checking controller inputs is easy. Just open the lid and click around through menus

dillysd, 2021-08-04:
Plot thickens; uninstalled and reinstalled on a new board PU-18 (since I have 30) Confirmed controller working but not on interface during boot. I am going to retest all the points again and see if I can get this pinned down to a bad board even though the controller works when the item was uninstalled. Ugh.