x-station/xstation-issues #141: Best way to create audio CD images (kinda dumb question)

badass-boss, 2021-08-04:
So here's the deal- as we all know PS1 is capable of playing audio CDs. However I have no idea how to create an audio CD image Xstation will recognize as I haven't done that for ages. The manual also doesn't state that in any way. What is the best way to convert any mp3/wav files to play on Xstation or use it as a swappable disc for other rhythm games?? Thanks for answering this lousy question

ramapcsx2, 2021-08-04:
Hm, I suppose you'd use Imgburn, and drag and drop wav files into the "Write Files and Folders to Disk" dialog.. Something like that :p

badass-boss, 2021-08-04:
Looks like I kinda answered my own question. Looks like ultraiso free edition does it just fine.
>>> https://www.trishtech.com/2016/03/how-to-create-audio-cd-image-from-mp3-files/

I realized audio CD tracks are really .wav files, however I remember ripping them from a real CD was not very easy and it required some software and copy pasting them in windows did not work. This got me confused. Also .wav files don't work on their own, however they work and play music in the player when I rename .wav files to .bin. This is interpreted as 1 music file in a CD though. There is still one question that emerges- does xstation support any other music or multimedia files, maybe right in the GUI?? Looks like playing music on PS1 is kinda easy on its own

blzla, 2021-08-04:
This has been discussed on Discord last year, without a great solution. UltraISO works and has been tested on XStation, but you have to pay attention during the installation because it will install adware by default. Disconnecting from the internet during setup is also a safe way to install it. Iirc the freeware version has severe limitations, so you'll need to register the full version for full length audio CDs.

badass-boss, 2021-08-04:
UltraISO works just fine to me and I don't see any limits. I even created a 900MB file which is out of audio CD specs and it STILL works on PS1. It works in the player and games and I believe all of the songs are playable. That is a nice bonus I didn't expect!

badass-boss, 2021-08-05:
I just loaded another audio CD image. This time it is 1.5GB and the BIOS player loads it but the counter and song switching is all messed up and the player picks up any song at any moment. I tried to play it from the start and leave the console on and to my surprise, all of the songs just played perfectly and everything looped to the beginning where it should be. From the technical point- how is this even working?? This is way beyond 700MB CD specs. Does this mean Witcher 3 with 50GB can be ported to PS1 and played on Xstation??

Getting back to the original question then it is pretty much solved. UltraISO free edition is the way to go and is very versatile. It takes any mp3 or wav files and converts everything straight into an audio CD image as expected. Every other common software I tried previously was garbage. This question is potentially solved. If anyone has even easier solution or any sort of script or ideas then they can also be discussed here

ramapcsx2, 2021-08-05:
Well, the PSX can address (seek to) 74:00:00 "minutes".
If the content has been started before that and left playing, it will play fine until any physical limit is reached.
With an ODE, the limit is arbitrary, and I think I have it clamped at some high value... It's odd if it played the entire 1.5GB :p

franzstation, 2022-07-05:
Hello everyone! I have the same question. Can u guide me to run audio CD images on the xStation?

I've tried too many ways, and I can't make it works! 😓

ramapcsx2, 2022-07-05:
Old issue, but the addressing limit turned out to be much higher, though it's not very precise.
Seeking above 90:00:00 seems to still work :)

closed 2022-07-05 by ramapcsx2

Enecororo, 2023-04-16:
Gave it a shot myself, I downloaded the NSP album Cool Patrol off of youtube music (though some were just the music video uploads that ended up converting to MP3s) and I turned them into a bin/cue file with UltraISO. Only real snag I've noticed is that the play all function on Vib Ribbon jumbles some of the audio tracks, playing the outro track right after the intro one