x-station/xstation-issues #142: Screen colour issue

Marki5legend, 2021-08-05:
Screen is very discoloured on my modern Samsung TV using the standard rca connection.
But works perfectly on my crt tv
[image]

ramapcsx2, 2021-08-05:
You are playing a game that is out of your region (I'm guessing an NTSC game on a PAL console).
The PSX can do this, but you will not be able to use common video cables, as the color will be off.

More info and workarounds here :)
https://github.com/x-station/xstation-issues/issues/122

Marki5legend, 2021-08-05:
No happens with all games and the main menu

#122<https://github.com/x-station/xstation-issues/issues/122>

ramapcsx2, 2021-08-05:
Try setting the loader to PAL mode in the options. Does it fix it?
(Also try setting it to NTSC if it doesn't.)

If none of the options fix it, your PSX video circuit might have problems.

Marki5legend, 2021-08-05:
I did try both those options. Might have to try another tv. Strange it works on my crt fine but not a flatscreen

ramapcsx2, 2021-08-05:
That could be explained by different tolerances. 
If you can, you may want to have the console checked out. Maybe a capacitor replacement is in order.

badass-boss, 2021-08-05:
That's an extremely common issue. PS1 was made in the 90s while your LCD TV was probably made in 2010 at the oldest so expect incompatibilities here and there. PS1 may be 100% fine. Consider using RGB SCART cable if your TV supports this input. Optionally consider using YPbPr component cable if your TV supports 240p or 288p and this would be a great solution overall. Component cables for PS1 are special however and aren't available everywhere. If you are interested then I can sell such a cable as I have parts but that's another topic

ramapcsx2, 2021-08-05:
That's the better recommendation. Thank you :)

closed 2021-08-12 by ramapcsx2