x-station/xstation-issues #143: Toasted x-station

jmoonb, 2021-08-10:
Pretty sure my x-station is dead.  Had my playstation on with the lid open and tweezers in hand.  A recipe for disaster.  Ended up shorting pins 2 and 3 causing everything to shutdown.  I'm guessing the esp32 doesn't like getting that much voltage.

Guess this is more of a warning then an issue.. unless there is some miraculous procedure here to reverse any damage.  Those pins are pretty close to each other so do take precaution!

ramapcsx2, 2021-08-10:
Pins 2 and 3 on the regulator IC? The one that has 3 pins and does the power?

If so, a couple different things can happen. It depends on which pins exactly got bridged.
Often you're lucky, and you bridge a supply pin with ground. A fuse in the PSX might be blown.
Does the green LED stay unlit now? It could be the PSX fuse then.

jmoonb, 2021-08-10:
No, pins 2 and 3 on the x-station board itself that connects the ffc.  From what I understand, 1 is ground, 2 is for power and 3 is for the connection to the cd controller.  I saw a brief spark and poof.  No power to the x-station.  The playstation does power up though but obviously no signal.

nicolasnoble, 2021-08-10:
Regulator ICs very easily blow up, but also are really simple to change. You might be lucky here  and able to change it.

ramapcsx2, 2021-08-10:
Well, if the green LED (on the xStation) comes on, then the LDO is working. That would be a bad sign.
If the green LED stays off, then first check the PSX fuse.

jmoonb, 2021-08-10:
I just opened up the console and tested the fuses and you were right!  2 blown fuses.  May have dodged a bullet.  I'll update you once I find replacement components.

badass-boss, 2021-08-12:
Many retro consoles are loaded with protection measures like fuses to prevent anything serious from happening. In this case yeah Playstation 1 consoles are loaded with lots of fuses and top quality protective measures during that time. The same cannot be said of Sony smartphones quality today

jmoonb, 2021-08-22:
Ok so I have an update.

Changing the fuse worked!!... at least for a little while.  I tested first with just the power and av cables but once I added in the controller port I ended up with a black screen.  There is output though.  I can see the video text disappear and 480i change to 240p (I use a CRT).  I can also see the blip on the screen once the x-station turns on but it never gets to the bios start screen.

I rechecked and replaced all fuses, tested/replaced capacitors along the power path but nothing.  The only discrepancy I found was a voltage difference on the points connected to pins 4,14 and 15 when the x-station is connected.  Voltage drops by half and to practically 0 on pin 4.  Thought it was an loose FFC issue but I get the same after soldering the tips.  Then again, this could be totally normal behavior.  

Not sure what I can try more from here.  Maybe the controller port shorted something and the console is borked.  I am open to suggestions.

ramapcsx2, 2021-08-23:
Never any video output (visible pixels drawn)? Check the APLL wire for shorts.

jmoonb, 2021-08-23:
> 
> 
> Never any video output (visible pixels drawn)? Check the APLL wire for shorts.

Nothing.  Just the subtle change from 480i to 240p and the blips on power on.  So there is output.. just black.  I checked everything for shorts but couldnt find one.  I'll have a look over again later.

jmoonb, 2021-08-24:
OK Final update.  After testing, re-flowing, etc with no change, I decided to just dunk the whole board into IPA and scrubbed it clean.  And it works!!!

closed 2021-08-24 by jmoonb

ramapcsx2, 2021-08-24:
Glad to hear :)