x-station/xstation-issues #144: Booted to CD player upon firmware update

Test045, 2021-08-11:
Not a real issue per sey. I have two xstation, one on a pal console and one on an ntsc-j console.
My ntsc-j console wasn't up to date (since early 2021 I would say) and I decided to update it.

After the update was done, it restarted as usual and I got back into the xstation menu. I refreshed the game list so that I could now have the newly added 240 test suite, as i wanted to recalibrate my crt geometry.

When i launched the 240p test suite, it loaded almost instantly into the cd player. I tried to hold reset (maybe not long enough) and it still booted back into the cd player. Powering off and back on the console led back to a perfectly working xstation. The 240p test suite now seems to be correctly loaded by the xstation everytime, instead of booting instantly into the cdrom player.

Any idea what could have caused this ? A weird, unlikely bug upon refreshing the game list that could have led to some sort of difference between the "model" and the "view" exposed by Xstation ? I've done the soldering job myself on both consoles and never had any issue whatsoever, with hundred of hours of gameplay on my ntsc-j one. I just want to make sure my work is likely not at fault here.

Test045, 2021-08-11:
Yea I can only imagine we'll probably never find the cause of this but I'm still interested in hypothesis. I tried to reproduce it for a good hour yesterday, and nothing. I wonder if the firmware I had was so old that it could have somehow caused this bug, given that all the files on the sd card created by the xstation were on previous firmware. I updated to the new firmware before updating the gamelist, for what's it's worth.

ramapcsx2, 2021-08-11:
Oh yeah, it's likely a problem with an ESP32 peripheral (such as a timer).
These have to be reset and initialized on power up and on firmware update related resets.
Possibly an older firmware used a peripheral, the update reset it, the new firmware uses it again but the state is not exactly right..
It's hard to debug this, so the good old turning it off and on again should clear it all up :)

Test045, 2021-08-11:
Aright then, I guess you can close this ?
I've been playing literally all day (I'm on holiday) and as the first day I soldered my QSB and powered up the xstation, I have absolutely no issue, so unless this issue magically rises again, I'm happy with the answer you've given me.

ramapcsx2, 2021-08-11:
Happy to share what I know about it :p

closed 2021-08-11 by ramapcsx2

calaerial, 2021-09-04:
This may be the same root cause as the issue im experiencing, except mine didnt go away when powered down :-S

ramapcsx2, 2021-09-05:
Nah, the small talk in the end was about a glitch that can happen after an update (should not, but it may), that will definitely go away on a power cycle.