x-station/xstation-issues #145: Ape Escape NTSC-J Redump ending cutscene doesn't play

BatteryPine, 2021-08-12:
I was finishing the Monkey Madness stage of Ape Escape and as I went through the last gate my game had a black screen until I reset my console. The audio for the cutscene was playing before it went back to the title screen music, but the screen was still black. I can provide my save data from my MemCard PRO if it helps with figuring out the exact issue.

ramapcsx2, 2021-08-12:
Hey,
Ape Escape was mentioned before, and that user also had a black screen with audio running.
Could you try this scene on the new beta firmware? It's on Discord here: https://discord.gg/kNK4UaR6

BatteryPine, 2021-08-14:
I tried the beta firmware and it worked!

ramapcsx2, 2021-08-14:
Okay, I'm closing the issue but I attached some labels. It might come up again, which would make this a good test game.

Fluoxi, 2021-11-08:
I'm having issues with Ape Escape aswell, 1.4.4 PAL. I'm at 7-1 Sushi Temple, black screen no sound when starting the level. Testing around first level is fine but when starting 5-3 Hot Springs I also get a black screen but with sound.

ramapcsx2, 2021-11-09:
Console and game are both PAL? If so, do you have a save game for me?

Fluoxi, 2021-11-09:
Both Pal, I don't have a way of grabbing my save files at the moment. For additional information I played the game up to the level I'm at then modded the console. I was able to get into the level before the mod. Already played several games with no issues. I'll go check some more levels.

ramapcsx2, 2021-11-09:
Well, it's good if the error is repeatable. Maybe I can cheat to that level.

Fluoxi, 2021-11-09:
I don't know what to make of it but every single level up to my save point works except 5-3 hot springs (black screen with sound) and 7-1 sushi temple (black screen no sound).

ramapcsx2, 2021-11-11:
Update: I've got a save file for level 7-1 now, which does indeed hang. On it!

ramapcsx2, 2021-11-11:
Update2: It is a problem with how I implemented libcrypt. 
It's not hard to fix, so please stay tuned for next release :)

Fluoxi, 2021-11-11:
Thanks for checking!

ramapcsx2, 2021-11-27:
Same problem with Parasite Eve 2 Italy, which is generally a weird LC title. But it happens there when trying to enter a garage mid-game.

Fluoxi, 2021-12-15:
Solved with the latest update, thanks!