x-station/xstation-issues #148: Lid switch issue (?)

Jadoube19, 2021-08-29:
I‘ve been experiencing a weird issue since I installed my xstation and PS1Digital. In normal operation, both devices work great (no stutters, crashes etc). The issue is that the console thinks that the lid is permanently closed, as if it was taped down (which - for the record - it is not!). Annoyingly, this renders games likes Metal Gear Solid virtually unplayable because you are required to swap discs without getting to save.
I thought I could just bypass this through the lid switch toggle on the PS1Digital, but whenever I enable the toggle via the web console, I end up on the memory card screen. I have tested the wiring and it seems to be fine.

Any ideas on what‘s causing this? Is it a faulty lid switch/bad contact on the mainboard or could there be a hardware-level conflict?

ramapcsx2, 2021-08-30:
Since you get to the Memory Card screen, the lid switch signal is able to report both states.
I don't know what kind of options the PS1Digital offers for the lid switch, but maybe one of them is interfering?

ramapcsx2, 2021-08-31:
Try option 2 and find the pad combination that is supposed to "virtually" open/close the lid :)

Jadoube19, 2021-08-31:
Great idea, but as far as I know the switch can only be used for swapping multi-disc games. First you have to choose the disc order in the X-station menu which I can‘t get into. Really appreciate your help!

ramapcsx2, 2021-08-31:
Yes, I'm assuming you don't know or didn't use the pad button combination that tells PS1Digital to close the lid.

Jadoube19, 2021-08-31:
It‘s actually indicated in the menu, so no difficulties there :) But no, nothing happens.

ramapcsx2, 2021-08-31:
Okay, that's an indicator that the soldering for that function isn't good. I think PS1Digital can't drag the signal to ground.

Jadoube19, 2021-09-01:
Allright, seems very plausible. I will check the connections and report back!

GizmoTheGreen, 2021-11-10:
didn't know about this function of the ps1digital, very cool.

I noticed you need to cut a trace too, not something you forgot maybe?