x-station/xstation-issues #149: Multidisk games will load on powerup instead of going to loader.

jmoonb, 2021-08-31:
So this one is a bit strange.  It only seems to happen with multidisk games in a particular scenario. 

What I would do is load a multidisk game with fast load.  Then after playing I turn the console off.  I then open the lid and power on the system which goes into the memory card/cd player screen.  I then close the lid and it loads the game again instead of going into the loader.  

I guess this wouldn't be a big issue if it worked but the game doesn't load properly and it asks to put in the right CD after loading a save file.  A reset will remedy this issue.  Tried the exact steps with single CD games but had no issues.

ramapcsx2, 2021-08-31:
On 1.3.0 or the latest beta? This was changed in the beta recently.

jmoonb, 2021-08-31:
1.3.0.  Good to know that this has been fixed though!

ramapcsx2, 2021-08-31:
I'll leave the issue open, in case the new version doesn't fix it :)