x-station/xstation-issues #150: All Star Racing 1 and 2 issue

gtsamour, 2021-09-01:
The games All Star Racing 1 and 2 (USA latest redump images) seem to have an issue with dual shock controllers (I have a PAL console).
There seem to be intermittent Start button presses without me doing anything, in menu and during game, which causes the game to pause. The problem appears with both fast load and regular load.

I tried a Dual shock PSOne controller and a dual shock PS2 controller and with both its the same thing. All of my controllers pass the Analog Controler testing program so they work fine, and with all the other games.

But when I use a non dual shock (old) PS1 controller both games behave right. I searched online and there are no issues reported with these games so it can't be a game bug.

ramapcsx2, 2021-12-03:
I'm sorry, I totally didn't see this issue.
The reply may be too late now, but have you learned more in the mean time?
If it happens in full and fast boot modes, then it's unlikely to be caused be the ODE.
There's no functionality regarding pads in the xStation hardware.

gtsamour, 2021-12-03:
Hi,

no I didn't find anything new, I just noticed the issue (in both modes, fast and full) and thought i would report it. I dont know if its the ODE or not or if these were early games that only supported the early non dual shock controller and have issues with the dual shock ones.

ramapcsx2, 2021-12-03:
It's probably an issue with these games then.