x-station/xstation-issues #153: Certain games not loading after logos

Danimaltl, 2021-09-04:
I have a SCPH-1001 system that is playing most games fine. However the screen stays black when trying to load certain games: Suikoden 2, FF7, FF8, Tony Hawk 2, and MLB 2001. 

This issue is present using multiple types of SD card and multiple ROMs with verified MD5s. The xstation installation looks fine and I was told on the Classic Gaming Discord that it could have a faulty DSP chip. Are there any recommendations for further troubleshooting?

[image]
[image]
[image]

ramapcsx2, 2021-09-05:
I don't think it's the DSP chip, as in that case nothing would boot.
Typical suspects are any SD extender cables or otherwise a weak contact between all the SD card pins and the xStation.
Another common fault is a cold solder joint on the QSB, where a contact is intermittent.
Also check where you lifted pins with a magnifier. Make sure that the pads that are left over after lifting don't accidentally short out (usually right below the chip plastic is a popular spot).

ramapcsx2, 2021-09-05:
The installation looks good from what I can tell from the pictures. Please verify this one though:
[image]

Danimaltl, 2021-09-06:
That joint is a little spikier than the others but looks fine. I checked the lifted pins as well.

For the SD connection the result is the exact same whether or not I use the SD extender. Pressing on the card does not have an effect either. The issue consistent whether or not I mess with the connection.

Video of booting the game w/ xstation lights https://photos.app.goo.gl/KwcxPE1hUYdfV6y5A

ramapcsx2, 2021-09-07:
The initial startup and all the loading there look spot on. There's no issue with the license symbols. The game .exe starts running fine, loads a bit of data and then it seems to die.
Okay, I'm at a loss on what this could be.
Which firmware is installed?

Danimaltl, 2021-09-07:
Yep, it hangs at the exact same spot. I've been chatting with the modder I bought it from and will be sending it back in exchange for a different unit. They'll be able to do more rigorous testing and hopefully figure out why it has this weird, specific behavior.

ramapcsx2, 2021-09-07:
Alright, best of luck that the replacement works!

bikingjahuty, 2021-09-11:
I am the installer on this specific PS1. 

Danimaltl returned the Xstation modded PS1 to me along with his SD card loaded with his disc image files. First thing I did was fire the PS1 up with his SD card and sure enough the games he mentioned didn't work (Suikoden II (USA), Tony Hawk Pro Skater 2 (USA), Final Fantasy VII Disc I (USA), and Final Fantasy VIII Disc 1 (USA)); the Sony and Playstation screens would pop up as they normally do, however you'd get a black screen after which it never progressed past. This happened with both Quickboot and Full Boot. I then used my known working SD card with known working files of the same games I tested out on Danimaltl's SD card, and sure enough I got the same behavior. Just to make sure nothing happened between the time I installed the mod and sent it to Danimaltl, I retested Tekken 3 (USA), Einhander (USA), and Castlevania Symphony of the Night (USA) on my SD card and all worked fine. This is consistent with what Danimaltl originally reported about certain games working fine, while others did not. 

I have a working, Xstation modded PS1 on hand that I tested the non working games off Danimaltl's SD card and they all worked perfectly on this PS1, as did the same games on my SD card. This ruled out issues with the game files or SD cards.

I then removed the Laserbear SD mount on Danimaltl's returned PS1, but still got the same non-working behavior from both SD cards and the problematic game files. 

I have a brand new, never used or installed Xstation on hand that I swapped into the returned PS1 (just the Xstation board) and after doing this all the game files on Danimaltl's SD card worked perfectly. I then tried my SD card on the swapped Xstation board inside the returned PS1 and the same games worked great. 

To rule out issues with corrupted firmware on the original Xstation board I both updated the firmware from 1.2.0 to 1.3.0, rolled back to 1.2.0, and then updated back to 1.3.0 again. Between these firmware updates, I was unable to get the problematic games to work, again on the original Xstation inside the returned PS1. 

And to be absolutely sure it wasn't anything else involving the install I confirmed continuity on the QSB solder joints, verified no continuity between lifted pins and any other pins, and also rechecked all removed resister sites. I could not find any faults with my installation as confirmed by my multimeter and visual inspection. 

I am 100% certain the issue is a faulty Xstation board given all the tests and troubleshooting I performed. 

ramapcsx2; I purchased the faulty board from Castlemania Games. Do I go through them for an exchange or is there another process I have to go through? Also, if there's anything else you'd like me to try I'd be happy to. I am sending Danimaltl another modded PS1 that works perfectly with his game files and SD card.

ramapcsx2, 2021-09-11:
Thank you for the thorough testing!
It's clear that there's a signal problem on this xStation board. That it only shows in certain games is unusual, but I suppose it can happen..

In any case, you should email Castlemania Games with the customer/order number. 
The board will be replaced, of course.

closed 2021-12-03 by ramapcsx2