x-station/xstation-issues #154: X-Station boot failure after update

calaerial, 2021-09-04:
Hi all, 

My X-Station has been installed for a number of months at this time, its been moderately used. Played through Parasite Eve and a few others in that time with no substantial issues. I decided to update the firmware this morning to get the folder browsing features and im now in a state of epic regret. 

Since the update the X-Station cannot boot, it was working until this point without issue, the firmware update seemed to go through without issue or error. It rebooted itself when it said it would. 

So far ive tried the following:

1: On cold boot the PSX simply boots directly to the Music Player screen, note not to the memory card/music player menu expected if the lid was open, so its not that. It goes directly to the music player as if it has detected a music CD. 

2: On a reset with the lid closed it will sometimes detect software in some way or another as it will go to the black SCEE logo as if booting a game. It then simply hangs on a black screen however. 

3: If the lid is open on boot it goes to the memory card/music player menu as expected, if the lid is then closed it will behave similar to a reset, it will detect the software in some small way, give the SCEE logo and then black screen.

4: Ive taken the PSX apart and tested every solder point using mmmonkey's test point diagram, all appear in order. So far as i can see there are no issues with the install. 

I am really worried that the update has killed it. Is there anything else i can try to get this back to life?

ramapcsx2, 2021-09-05:
The problem is certainly with the PSNee license symbol injection. I suspect that there is a soldering issue (weak solder joint) on the QSB, either where one signal should be grounded, or the license signal point itself.
What happens is that sometimes (but rarely), the symbols go through correctly, but it's not reliable enough to boot any software.
Your QSB has soldered down connections at the points marked "Data" and "Gate" in this picture:
https://cdn.quade.co/wp-content/uploads/2018/11/PU-18-Diagram.jpg
One of them isn't making a good connection to the xStation.
The soldering could be bad, or the white flat flex cable could have a bad connection.
Try to replug the flat flex cable (maybe clean the contacts as well) a few times.

ramapcsx2, 2021-09-05:
As an explanation on why it worked before the update and now not anymore:
The slight timing differences can shift what the mechacon chip still accepts as valid signals, versus what it discards.
It's not the update that broke your installation. The installation had a hidden fault all the time.
Knowing this, you should be able to fix it with the recommendations above, hopefully :)

ramapcsx2, 2021-12-03:
Any luck?

calaerial, 2021-12-23:
Hi Rama, 
Thank you for keeping an eye on this so long after it was posted and for querying progress.
The X-station started working after this and has been working flawlessly since.

I couldnt really account for why, all i can guess is perhaps some issues with the memory card. If it reoccurs I'll try another.

Again, thanks for the support on this :)

closed 2021-12-23 by calaerial