x-station/xstation-issues #155: Booting to BIOS/Memory Card. Blue Power Light and Green Light Blinks Twice Before Shutting Off.

kemingruu, 2021-09-10:
Hello, 

I've recently got an xstation and have followed the install process. I have no idea what I'm doing wrong, as it seems all the solder points are good, and the pins are properly lifted. It boots into the memory card bios and I notices it has a blue light and the green light blinks twice before shutting off. When I don't put in the memory card, it continuously blinks until it eventually boots up into the memory card bios. Also, there is no sound with the Playstation logo boot up. 

It was able to play CDs and run games pretty smoothly, although the laser was starting to die, hence the replacement. I did test another SD card as well, and it does not work. I don't have any PS1Digital installed, just the xstation as my first mod. 
[image]
[image]
[image]

Here are some pictures of the installation. Maybe I'm being dumb? Any help is appreciated.

ramapcsx2, 2021-09-10:
Hey,
the problem description points to a bad QSB installation. One of the points doesn't make contact, or points are shorted.
If this is your first modification, I recommend you get this looked over by a modder :)

ramapcsx2, 2021-09-10:
Right, and now I've read the chip number of the chip you lifted the pins. Do you see the problem? :)

retrofank, 2021-09-10:
Yep. Sorry dude, you made a boob error and lifted the pins on the wrong chip.  Plus, looking at the pin 47, it's touching pin 48 by the looks of it, so that would be an issue, if it was on the correct chip anyway.

GitControlDel, 2022-02-21:
Another cause for this that I can't stress enough for anyone else with this issue, especially those with a PS1 Digital, is the lid sensor, you can see it's status using UniROM and telling it to load the disc (No disc or optical drive required), it will display if the lid is open or not. You can take the lid off and hold it down to see if it solves your issue. My system refused to boot into xstation unless I put my all my weight onto the top case with my hands. I suspect the PS1 Digital is to blame with it's added height. Need to 3D Print something to help extend the tray leg or maybe use a dab of hot glue.