x-station/xstation-issues #156: Wrong button listed on first boot

L10N37, 2021-09-24:
Nothing major, just the Xstation tells you to press triangle instead of circle ⭕  on first boot 

Triangle doesn't do anything 
I thought I had a blown controller fuse on my last install.

ramapcsx2, 2021-09-24:
Hm, with a blank configuration file and no cached game list, the loader will tell you to press triangle to enter the options menu.
In that menu, you are able to scan for games or switch to folder browsing.
One of these is required to be able to select any games :p

L10N37, 2021-09-24:
Yeah but triangle doesn't do anything. You have to actually press circle

ramapcsx2, 2021-09-25:
Can you tell me which firmware and loader.bin were used? I don't remember ever running into such an issue.

L10N37, 2021-09-25:
It's the latest firmware, the one without a multi disc Q. Ive got one in my personal collection so I can replicate it and grab a link tomorrow on video. 

It was a few weeks ago, all I remember is on 2 consoles in a row it was telling me to press triangle, but that button didn't do anything, where as circle did .

L10N37, 2021-09-26:
I didn't get around to this as it's really such a minor issue, I finally had myself a rest from console modding and been playing through parasite eve on my super sexy modded 5502.  I'm onto the 'insert disc 2" screen but the latest FW doesn't have the multi disc queue. How do I switch discs?

L10N37, 2021-09-26:
Also Rama, i have a dual XTAL solution that differs from the DFO

https://github.com/L10N37/PSXTAL

I saw someone complaining about NTSC games on his PAL console. I offer the PAL60 mod which is obviously the old school PAL colour burst for 60hz with PAL colours but I thought you maybe interested in this. 

You can switch between each XTAL by holding L1 + R1 + circle for about 4 seconds on all genuine controllers in digital mode. Its open source and also contains short and long reset button combos.

ramapcsx2, 2021-09-26:
Folder browsing is new in firmware + loader 1.3.0. The disk queue is not yet implemented there. It will be in the upcoming firmware, so stay tuned for that :)
In the meantime, you can switch to list view mode in the options. It supports multi disk queues (select disks with Square).

I see what you're doing with the clock switcher mod. It's a working solution, but let me suggest you upgrade to an ESP32 and use its integrated APLL.
You can simply generate the required clocks, at a good enough quality to satisfy the needs for composite video encoding.
You won't need difficult to source oscillators then anymore, it can also interface to the pads, plus a bunch of other possibilities (add a modchip!).
This all for about the same cost as just the Arduino :)

ramapcsx2, 2021-09-26:
I just put the 1.3.0 loader.bin on my SD card and deleted all configuration files. This is what I get on startup:
https://i.imgur.com/UtVVQCC.jpg
Then I press Triangle and get:
https://i.imgur.com/aq6yO5S.jpg

L10N37, 2021-09-26:
Thanks
I actually ordered an ESP32 a few nights ago.
The large crystals are easy to source, but obviously an ESP Is a better idea, although the ESP is 10/11AUD and the nanos a couple bucks a pop. It's a CH340 clone. 

Yeah that's the screen but I definitely had to use circle, on 3 systems! I will look into it and get back to you.

Also, I found another early PU8 revision not mentioned in the guide. The boards floating around somewhere, I'll grab the number off today.

L10N37, 2021-09-26:
Hang on... Are you pulling my leg? That says the FW was released yesterday. That's like 10 hours ago lol

L10N37, 2021-09-26:
http://imgur.com/gallery/SqVbWP6

Other PU8 

Cheers Rama
Thanks again for this ODE.

 10/10

ramapcsx2, 2021-09-27:
The firmware is in beta, and the screenshot is of the development version. But the firmware doesn't affect the problem, as it's all about the loader software.

L10N37, 2021-09-27:
I put the SD card on the PC and wiped the Configs and it's working fine. Went through all controller types and triangle works. My bad. Apologies for the false alarm. 

Keep up the good work :)

closed 2021-09-27 by L10N37