x-station/xstation-issues #157: No controller inputs after install.

BK2K, 2021-09-24:
System appears to boot the xstation just fine, it boots and gets to the point where it says no games found and to press triangle only I can not pass this point as no inputs work.

PU-18-52 pcb install that went well.
Working system but for drive read issues before install.
The test points posted all check out for the QSB.
Controllers all test ok different system, all fuses test good. Connection from control port pcb to pu18 pcb all tone out. (ribbon cable ok)
Connection from CN102 to cpu IC103 looks ok best I can tell. (repair manual found online is not the best)

At this point I am at a loss as to what could be happening. In searching google I found a few posts with same issue but no resolution. I am hoping this is possibly a known issue or at least some ideas of what else to test. What with the xstation can affect the controller system? 
I dont want to uninstall the QSB if I dont have to, but is there anything under it that could short or cause this issue being in contact with the QSB?
Thank you.

jmoonb, 2021-09-25:
Try doing a thorough cleaning and drying of the board. Cant stress how many times i brought things back to life just by cleaning. I once had issues due to some flux and ipa getting stuck under the components and cpu pins near the controller ribbon connector. Wouldn’t work at all at first but would slowly come back after a day erratically until becoming fully functional again.

Of coarse this assumes everything else was tested and working such as the memory card, voltages, continuity, etc.

ramapcsx2, 2021-09-25:
This is good advice. In addition, I would recommend to open the lid and start the console into the BIOS menu. Do the pads work there?

L10N37, 2021-09-26:
Did you try circle instead of triangle? I just lodged an issue for this the other day. I thought I had a blown fuse. I switched the controller board out and checked the fuse and everything cos it says triangle but that doesn't do anything

BK2K, 2021-09-26:
No there is no input anywhere, bios with lid open or at press triangle. And yes I mashed away at all the buttons as any upset desperate person would.
I pulled the metal shielding over the cpu to better check connections from CN102 to the cpu and to make sure its clean.
My question is still what part of the install process or what of the xstation itself, or its connections is capable of disabling or killing the inputs like this.

ramapcsx2, 2021-09-26:
None, basically. The install taps into 8V that is protected by a filter and the 8V line fuse towards the CD system.
The QSB connects to CD system points, all very far removed from the CPU controller port side.

Total loss of input, hmm. A bit hard to tell then whether Memory Cards work.. Dunno :/

BK2K, 2021-09-26:
Well bummer. I was feeling hopeless but a buddy is going to give me a working 1001 with a pu-8-23 that I will give a go on and see  how it does. If this one works out I will probably try to revert the pu18 back to stock to see if I can bring the inputs back to try and learn what happened. 
The resistor that was removed from the pu18 is lost, do you know the value? Is it a 220 ohm?
Thank you.

ramapcsx2, 2021-09-26:
I'm going off memory here but yes, I think that is the correct value.

BK2K, 2021-09-30:
Ok so I uninstalled this in the 5501, nothing weird under the qsb. It returned to normal audio and cd drive booting but still no input so must be mostly unrelated to the install. Probably some issue from board flex when opened up that I dont care to hunt down.
Did the 1001 install and it is running fine after a few hours of testing.
I guess the only remaining concern I have is this 1001 gets crazy hot. I know that is "normal" for the 1001 and its power supply but its hot enough I dont want to run it like this until I figure some way to cool it down. The xstation itself also gets hotter than I would like it to. Any thoughts about a heatsink on the esp or...
I was looking at the 2 replacement power supplies that use the 12v power but what I was seeing it sounds like they are not well built.

ramapcsx2, 2021-12-11:
Hm, my update notifications seem to miss some issue updates.. Sorry for that!
The PU-8 installs do get warmer, but it's all still well within margin. Especially the xStation itself is definitely fine. There's no need to install heatsinks.