x-station/xstation-issues #158: Destruction Derby 2 (NTSC) Garbled graphics when races start

8bitgamer88, 2021-09-27:
When I start destruction derby 2, the start up logos and main menu work fine but once it goes into a race the graphics are all messed up

ramapcsx2, 2021-09-27:
The problem is known, but I thought I had it fixed for quite some time. I can't reproduce it anymore on none of my machines.
Even when it was reproducible, it happened randomly on loading a level. It could take an hour to happen.

Which firmware are you using?

8bitgamer88, 2021-09-27:
1.3 and my PS1 model is a 5501 if that matters

8bitgamer88, 2021-09-27:
I've tried ripping from 2 different disks and even downloaded the rom from the internet. Maybe I'll try downloading the patch again and replace it incase something went wrong the first time?

ramapcsx2, 2021-09-27:
1.3 and on an NTSC-U PU-18 console, got it.
You are using some kind of patch? If so, does the problem go away if you leave the image unpatched?

8bitgamer88, 2021-09-27:
No patch, just straight from the disk

8bitgamer88, 2021-09-27:
Oh sorry when I mentioned patch earlier I was talking about the 1.3 firmware revision for the xstation. Not the game

ramapcsx2, 2021-09-27:
Update: This particular bug was still present in 1.3.0. The next firmware will at least reduce the frequency of it happening.
(I thought it was fully fixed, but it still happens on some consoles, after some hours of leaving the demo cycle run.)

8bitgamer88, 2021-09-27:
Awesome! Thank you for letting me know!

ramapcsx2, 2021-10-01:
Looks like it'll be fully fixed in 1.4.4 :)

ramapcsx2, 2021-12-11:
Haven't seen this anymore :)

closed 2021-12-11 by ramapcsx2