x-station/xstation-issues #160: Was working fine. Now boots into Memory Card/CD Player

arekuominae, 2021-09-30:
Hi,

So I have been using the xstation for about three weeks now, had a full play though of Castlevania SotN .  However as of this morning it just will not boot into the xstation menu and only goes to Memory Card/CD Player menu.

Nothing has moved or changed in hardware setup from last night when it was playing fine.  I can see that the xstation blue LED is on and the green light intermittently flashes before getting to Memory Card/CD Player menu.

I've tried another SD card and still will not boot to the xstation menu.

Any suggestions/thoughts?

ramapcsx2, 2021-09-30:
One of the signals is not connected anymore. Check the QSB soldering (or have it checked).

arekuominae, 2021-09-30:
> One of the signals is not connected anymore. Check the QSB soldering (or have it checked).

Appreciate the quick response.  Per your suggestions I reflowed all the points on the QSB and it has come back to life.  Not sure why it would have been working fine all this time, but happy it was a simple fix.

Thanks for your support.

closed 2021-09-30 by arekuominae