x-station/xstation-issues #169: Hanging/fmv stutter only when "cold"

Rubberdug, 2021-11-09:
Fired the console up for the first time in a couple months the other day and it first went to the bios screen for around 10 seconds then booted to loader as normal, tried it this morning and it stayed in the bios for around 30 seconds until I hit reset then it was fine, also getting stutter on tomb raider 2 intro (but not 3 or revelations), after it's been on for 5 mins or so all issues disappear until the system has fully cooled down again, everything was running fine before only difference  being the weather, tried reverting to a previous firmware but didn't make any difference, any ideas on why this would happen/how to fix? Cheers

ramapcsx2, 2021-11-09:
Hey,
this sounds like it could be anything, but it would be related to the installation.
If you had the xStation installed, could you check with the installer?
If you installed it yourself, the first thing to try is reflowing the QSB.

Rubberdug, 2021-11-10:
Gave it a reflow but its not made a difference, I'll try another board tonight or tomorrow, cheers