x-station/xstation-issues #170: wrong button shown

L10N37, 2021-11-16:
I found that button thing again , its a new 400gb SD card and when Xstation launches it says "folder empty, press O to go back. O doesn't do anything , triangle does.

ramapcsx2, 2021-11-16:
Do what fixed it last time :)
It was definitely not a button issue, iirc.

L10N37, 2021-11-16:
I couldn't remember exactly what the issue was last time as i reported it a couple weeks after, but I was able to reproduce it today on a brand new card. I made a video of it. maybe Xstations are getting shipped with an old version of the FW that doesn't support folder browsing or something. it's just that it says press circle, but the correct button mapped to go back is actually triangle.

ramapcsx2, 2021-11-16:
Ah yea, the devices ship with firmware that doesn't understand folder browsing or anything new at all.
It is expected that the user updates the firmware right away, and indeed, uses triangle to enter the options menu to do that.
I suppose this is confusing a bit.. Thing is, the shipping firmware is certified and hence I'm a bit reluctant to change it..

L10N37, 2021-11-16:
all good
 I was a bit embarrassed because it seemed like I made up a story with the last post, I recalled having to press a button that was different than what the screen prompted was all. No worries.. not exactly a bug, More a typo. It just threw me right off last time thinking I had a fried controller port .