x-station/xstation-issues #171: Cheat engine compilation

sigarette1978, 2021-11-23:
WHEN I PUT THE CODES ON ACTION REPLAY 2 V2 THE GAME DOES NOT START THE PLAYSTATION BLOCKS

ramapcsx2, 2021-11-27:
Hey, 
does the game run if you don't put in any cheats?

sigarette1978, 2021-11-30:
IF YOU INSERT THE TRICKS THE GAME BLOCKS, TRY IT

ramapcsx2, 2021-12-01:
So I assume you're making a mistake with the cheats or the way you want to start the real game then.
Please try a different cheat disc.
I know this stuff works, tried it before :)

sigarette1978, 2021-12-01:
try with ACTION REPLAY 2 V2 adding codes to any game causes the screen to freeze
Firmware + Loader [1.4.4]

sigarette1978, 2021-12-12:
XStation Firmware + Loader [1.4.5]+ ACTION REPLAY 2 V2( engine compilation) freeze [https://www.youtube.com/watch?v=pKlfFr6v4SY](url)

ramapcsx2, 2021-12-12:
Please try removing some of the extra layers. Don't use the cheat engine compilation CD, but an image of that Action Replay 2 V2 CD directly.

Another thing to try is full booting the queue. I assume the console freezes due to one of the code overlays.