x-station/xstation-issues #172: Parasite Eve II hangs

carmax91, 2021-11-27:
Hi!

Playing with Parasite Eve 2 (ITA)  Disc 1 [SLES-02562]: If i try to enter in the garage location the game hang after the shutter sound.
I've attached a save and a image with the room where the crash happens. To reach the room you have just to go out from the balcony and take the ladder to appear in the "Water Tower" room, then you have to follow the arrow I drew on the map and it should crash as if you try to enter in the "Garage" room.  

The CRC of my dump is identical to the one of redump and I tried with other sources (both patched and not), but nothing to do, the game always crashes... Instead same dump work like a charm on the emulators.
Using the latest 1.4.4 update Fast and Full boot...
Hopefully i was clear!

Save and map attached:
[PE2 crash.zip](https://github.com/x-station/xstation-issues/files/7611885/PE2.crash.zip)

ramapcsx2, 2021-11-27:
Yep, thanks for the report and save file. I'll check it out asap :)

ramapcsx2, 2021-11-27:
I'm sorry for the inconvenience. This is the same problem as the other user had a week ago: I apply the libcrypt patch in a way that these games don't like. It'll be fixed next firmware!

carmax91, 2021-11-27:
Good!

So even with applied patch the xstation patching system always kick in... Maybe, if it is easly doable, will be a good idea to make an option to disable autopatching system for better compatibily for cases like this. What do you think?

ramapcsx2, 2021-11-28:
Nope. The patching is very clear cut and detectable on whether it has to patch or not.
It's the actual patching that had to be fixed, which I'll do with the next update.
Once this is fixed, it will not show up again in any other form either.

carmax91, 2021-11-28:
OK!
Bypassed the point (using emulator) and now the game hangs at first loading of the CD 2.
Looking forward to the fix, I hope it doesn't take long mate! 

Attached the save in case you need the test: Start with CD 2, you have only to go the garage room again and talk with Kyle and select yes (Si), after the cuscene the game hangs in the loading screen.
[Parasite Eve save 2.zip](https://github.com/x-station/xstation-issues/files/7614513/Parasite.Eve.save.2.zip)
)

ramapcsx2, 2021-11-29:
I'd like to bundle a loader fix with the update, but it shouldn't take too long :)

L10N37, 2021-12-03:
are US saves interchangeable with EU? I was confused about this issue as I'm a few hours into the game and passed in and out of them rollers multiple times.. no issues.. but I'm playing the SLUS version ;)

ramapcsx2, 2021-12-03:
It's only affecting the PAL versions, and within them, only those that happen to place data sectors besides LC sectors.

ramapcsx2, 2021-12-11:
Fixed in release 1.4.5 :)

closed 2021-12-11 by ramapcsx2