x-station/xstation-issues #173: region specific FW

L10N37, 2021-12-03:
Hi Rama, just hoping for region specific FW that only injects the magic key required. I assume Xstation does SCEI/SCEE/SCEA. I've had my own personal console become a bit funny with booting (since last FW update??) and now a customer complaint with the same issue (also latest FW, no issues prior) 

Customer console starts launching into MC card manager constantly after about an hour of gameplay. In game reset works, but takes him back to MC manager as well. He says once he leaves the console alone a while, it returns to normal.

for me, it was the opposite. a cold system would almost guarantee not to launch, or get to MC manager then suddenly show the black screen and decide to launch late etc. once warm, never fails. Final note ..playing Parasite Eve 2 (US version) on latest FW seems to break IGR modes. both from a mod and physical presses. Doesn't want to return to menu for a 2 second + push. ALWAYS reboots into Parasite Eve 2! 

no further testing, just some normal user experiences. 

thanks man :)

ramapcsx2, 2021-12-03:
Hm, I will have to replicate the license stuff. It is not so easy to solve by simply doing a single region, as the process is a whole.
What I assume is that all license symbols are mangled. This is just from experience with the system.
Can you give me details about the consoles that show this?
Do you think it's those consoles in particular, and not the entire series of them? (IE from weak capacitors or some such variation)

The issue with resets can not be caused by any software. It is purely hardware based.
Can you check that a bit more, maybe with a console without any extra mods on the reset line?

Cheers :)

L10N37, 2021-12-03:
the reset line is sent low at boot by a microcontroller , but it's a single line instruction and the next line basically high z's it. Probably unnecessary but basically ground out every port to be 100% there's no floating inputs to pick up noise. it's a fully recapped 5502 including PSU. The other is a 5502 as well, but I've put this in a few consoles now. 6 in total. Only my personal system and this one customer are having issues. It never happened prior to the latest FW update.

ramapcsx2, 2021-12-03:
Okay, so we have:
- License failure (maybe) depending on system run time or system temperature, exact variables unknown
- Some other issue with ingame resets, but only in PE2 (most basic game type, single track .bin, no LC)

I can definitely say that resetting via long press should work, regardless of game played.
The usual things around the reset line should be checked: PSU reset switch is clean, no other mods on the reset line, or such mod functions disabled.

For the license stuff, I can look over the code, but I'd have to have an example here to replicate the issue.
My test consoles all work fine, so this is tricky. Maybe I find something..

ramapcsx2, 2021-12-11:
Please give the new release a go. A timer fix might actually affect this, in a long roundabout way. Maybe 1.4.5 helps!