x-station/xstation-issues #174: Might have bricked the ODE while firmware update

super11114, 2021-12-11:
I'm new to the xstation world, but not new to modding in general.
Since QSB's for both supported versions were included, i wanted them both installed; i have some spare consoles lying around and didn't want to unpack my tools twice. 
I did the PU-18 first and did a quick check on if it boots - yes it did. So i installed the next QSB (PU-8-23) - it booted fine, as well. 
I used my finger to "close" the lid. Booted the system while finger was on. 
Released finger and looked around the menu and tried the "update firmware" option first.
It prompted me with "Is the lid closed?" or something - so i put my finger back on and it began updating (should have been v1.4.4) on its own.
After 3, maybe 5 seconds the screen went bad. Meaning, i had bars of blue and black running top to bottom on screen, no text. So i had my finger for maybe 10 more seconds on the lid before turning off the console just to make sure it wasn't part of the procedure, as the video was obviously frozen.

Now i get no video at all on neither console, complete darkness. Only the blue light is turning on. Even if i take out my SD Card, the green light doesn't blink anymore. Maybe my finger wasn't tight enough on the lid button and the console "opened" for a split second while updating or something? Would this have been enough to brick the main board? I can't think of anything else, assuming it's no faulty unit.

ramapcsx2, 2021-12-11:
That's very weird. 
The update process is "atomic". If anything goes wrong while it updates, it simply won't use the new code.
When you repower the machine, it will still use the unmodified old version.

The issue might be something else.

What happens if you remove the SD card and power on the machine?
(What happens on the green LED on the xStation?)

super11114, 2021-12-12:
The green LED is basically dead. It doesn't turn on, no matter if a SD Card is inserted or not. All i get is the blue LED. I imagine if this was caused by a faulty installation, at least one of the consoles would still boot.

ramapcsx2, 2021-12-12:
The ESP32 is fried, yes.

ramapcsx2, 2021-12-12:
I need a bit more info: Is there not even a small blip from the green LED on system power on?

super11114, 2021-12-12:
I tested both consoles now in a pitch black room. When the SD Card is inserted the green LED indeed turns on once for a nanosecond as soon as i power the console (before things went bad it blinked twice and a little longer irrc). It does nothing when resetting the console or pushing the lid button. Without a SD Card however the green LED does absolutely nothing, not even when powering on (before it used to blink constantly).

ramapcsx2, 2021-12-12:
Could you hop on Discord so we can chat? I think I have one thing to try :p 
https://discord.gg/jvyqwzfX

ramapcsx2, 2021-12-16:
So this xStation had an ESP32 module with a flaky SPI chip most likely. It would sometimes work, but freeze eventually.
Swapped the module, all is fine again.
The firmware update routine worked as it should :)

closed 2022-03-16 by ramapcsx2