x-station/xstation-issues #175: Tekken 3 (SLUS-00402) hangs on black screen on credits

lordairivis, 2021-12-11:
PlayStation model: SCPH-1001 NTSC-U
xStation firmware version: 1.4.5
Game: Tekken 3 (SLUS-00402), Redump image where Tekken 3 (USA) (Track 1).bin CRC32 = 8131AF42

Game hangs on a black screen after the ending cinematic plays, just after the voice says "The King Of Iron Fist Tournament: Enter The Tekken". No credits or additional music plays. Same issue on firmware 1.4.4.

ramapcsx2, 2021-12-11:
Okay, I'll check it out.

ramapcsx2, 2021-12-11:
Works fine here. Do you have the audio tracks + .cue file in your game folder? 
That would be something like Tekken 3 (USA) (Track 2).bin and Tekken 3 (USA) (Track 3).bin, as well as the .cue.

lordairivis, 2021-12-11:
Yep, cue file is as follows:
```
FILE "Tekken 3 (USA) (Track 1).bin" BINARY
  TRACK 01 MODE2/2352
    INDEX 01 00:00:00
FILE "Tekken 3 (USA) (Track 2).bin" BINARY
  TRACK 02 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tekken 3 (USA) (Track 3).bin" BINARY
  TRACK 03 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
```
CRC32s of tracks 2 and 3 are 28D8F221 and 662FEA9F respectively.

Using a SanDisk Ultra Class 10 128GB SDXC card formatted as FAT32. I can try this same image in an emulator later on, if it's helpful to you.

ramapcsx2, 2021-12-11:
Hm, I don't know... What happens is that the game is supposed to play an audio track then.
Do other CDDA using games work? A good test is Tomb Raider 1 or 2.

lordairivis, 2021-12-11:
Tomb Raider 1 & 2 both hang at a black screen right after I try to start a new game and no music plays at the main menu, but both play fine in PCSX ReARMed.

Tekken 3 also hangs in the same place in PCSX ReARMed, so it may be a faulty image in that case in addition to the CDDA audio streaming issue I appear to have.

I will need to try a different SD card I guess. Thanks!

ramapcsx2, 2021-12-12:
Hm, it's probably not the sd card, but somehow CDDA isn't working for you. This is pretty odd: Running games fine but not CDDA is untypical.
Best of luck debugging!

lordairivis, 2021-12-12:
Yeah, not sure what's going on. I tried a different SD card, same issue. I tried continuity testing the QSB and everything checks out, and I made sure to install some kapton tape beneath the lifted pins to make doubly sure there wasn't any unwanted connection, and still the same issue. I thought maybe the ribbon cables were not seated properly so I made sure they were seated fully and locked into the ZIF connectors. Not sure where to go from here to be honest.

ramapcsx2, 2021-12-12:
Castlevania uses XA, which is data being read in drive X2 mode usually.
Your issue won't be with the xStation install at all, otherwise you wouldn't be able to run games.

lordairivis, 2021-12-12:
You're right, of course. I'll see what I can dig up. Appreciate all the work you've done to bring this project to life!

closed 2021-12-12 by lordairivis

ramapcsx2, 2021-12-12:
Okay, this is probably something with the file system, or folder layout.. Something that would make a program not work with cue files and extra tracks. Good luck! :)

birdybro, 2021-12-18:
> 

Probably a dumb question but, have you tried adding a newline at the end of your cuesheet? The redump cuesheet is 12 lines, what you copy pasted was 11 lines. Just making sure.

lordairivis, 2022-01-29:
> > 
> 
> Probably a dumb question, but have you tried adding a newline at the end of your cuesheet? The redump cuesheet is 12 lines, what you copy pasted was 11 lines. Just making sure.

No such thing as a dumb question :). I went through 4 Playstations, uninstalling, reinstalling, trying to swap the chips that handle CDDA audio, the works. I remembered your comment and thought "fuck it, what the hell" and generated new cue sheets with the [DuckStation Cue Sheet Generator](https://www.duckstation.org/cue-maker/) for the problematic games and wouldn't you know it: problem solved. I feel incredibly stupid, but my own fault for not trying that first.

ramapcsx2, 2022-01-29:
Oh no, really?
So what went wrong initially.. did the CD tool mess up?

lordairivis, 2022-01-29:
Can't say for sure, I might have edited the cue sheets for a few of the games I was having issues with and messed up somewhere, or maybe the character encoding in my text editor was set incorrectly. Some of these games were found without cue sheets if I remember right, so I made my own by hand and I don't know if I made a mistake somewhere or if my text editor was set to ANSI or ASCII or something else.

Either way, it works now.