x-station/xstation-issues #179: xstation struggles to boot games and get past logos

VirtuaTom, 2021-12-17:
hello recently purchased a modded playstation 
I desperately need some help figuring out a problem

basically the console is very unresponsive sometimes it'll boot  x station menu then select a game and then it freezes.
Then if i restart the console it might not even boot the x station menu at all and will either freeze on the black playstation logo or the sony computer entertainment one, and then if i power up again it will just go straight to playstation cd menu......

I have noticed if the console is left alone for a while  it will actually boot the games to varying success but being switched off and powered back on seems to make it less likely to load right which is something i cant get my head around.
I really am puzzled what could be causing it work ok and then struggle to even get past the logo screens?

birdybro, 2021-12-18:
Sounds sort of like cold solder joints to me. Cold solder joints often have microcracks which can harm continuity, and when the system heats up a bit it can make the problem go away sometimes.

VirtuaTom, 2021-12-18:
I haven't had a response from the person yet, although i do have a few theories what it could also be
sd card
voltage problem
lid but i don't think it is that
bad solder joints like you said 
i think it's just a matter of narrowing it down, i just find it strange why it doesn't boot sometimes but then magically does other times.....

ramapcsx2, 2021-12-18:
Yea, that sounds like an intermediate contact, such as a QSB cold solder joint or also oxidized flat flex contacts.
You can try thoroughly cleaning the flat flex. If that doesn't change anything, look into reflowing the QSB.

VirtuaTom, 2021-12-18:
so small update i updated the firmware and xstation will boot almost every time i power the console on, however booting games is still not working sometimes I'm lucky and they'll boot and get past a couple warning screens and logos but then freeze again...could this be an sd card issue? or is it more likely that something on the install isn't quite making a connection 
I mention sd card because i have seen other people have had that problem if they aren't good quality

ramapcsx2, 2021-12-18:
An SD card issue is unlikely here, but possible. If you can run Ridge Racer Type 4 and play a round, the music should not skip.
If the music skips, that would hint at an SD card issue. If not, the problem is most likely elsewhere.