x-station/xstation-issues #180: Pepsiman NTCS sound problems at level 3 and 6

miguelk5, 2021-12-31:
When a giant rock ball or a Truck start to following you, the sound stuck when they say "PEPSIMAAN"

I don't know the technical implications and maybe this issue can fix others.

Regards.

https://user-images.githubusercontent.com/25555655/147825244-2ce21f7a-0b5c-49f7-b74e-724d8d05a482.mp4

ramapcsx2, 2021-12-31:
Hehe, this is the 2nd time I hear of this. It's not a bug, the soundtrack is this way :)
The NTSC-J and NTSC-U versions of the game have slightly modified versions of the track. You can compare them on youtube :p

miguelk5, 2021-12-31:
> Hehe, this is the 2nd time I hear of this. It's not a bug, the soundtrack is this way :) The NTSC-J and NTSC-U versions of the game have slightly modified versions of the track. You can compare them on youtube :p

Dammit we have been trolled, okey lets close the issue hehe sorry :)

ramapcsx2, 2021-12-31:
It does sound super glitchy :D

closed 2021-12-31 by ramapcsx2