x-station/xstation-issues #181: Added 1.4.5 loader and update files to microSD card, but now can't get past "Please Insert Playstation CD-ROM"

dritzthevampyr, 2022-01-02:
XStation was working fine, and I realized I hadn't updated it in awhile (maybe not since 1.2.0).  Decided to start the process to update it tonight and move the 1.4.5 loader and update bin files to the microSD card, but I cannot even get past the "Please Insert Playstation CD-ROM" screen.  I've held the reset button down multiple times.  I've done full resets through the PS1Digital.  The PS1Digital is even up-to-date (version 1.6.8).  I even took a spare microSD, formatted it, created the 00xstation folder path, and put the loader and update bin files on it only.  Same issue.

ramapcsx2, 2022-01-02:
Hey,
the files didn't cause any change. Something else broke, for some reason.

ps1lover, 2022-01-02:
hey games like Monkey Hero Formula One 99 2000 2001 by Studio 33 on 
xstation they are  working??? I am sorry if I am not helping here but I want to buy xstation  and these  games were never work good in emulator so I want to know if this games work in ODE and thanks

dritzthevampyr, 2022-01-16:
@ramapcsx2 Finally got around to opening up my PS1 and followed your suggestion from [this issue](https://github.com/x-station/xstation-issues/issues/118).  I reflowed solder to the * and 3 points, and now xStation boots up fine.  I even updated the firmware to 1.4.5 and loads fine.  I will keep an eye on it in the meantime.

I also noticed you added the snow background to the menu.  I really like it!!  :-)  Anyway, thanks!

ramapcsx2, 2022-01-17:
Good to hear, enjoy :)

closed 2022-01-17 by ramapcsx2