x-station/xstation-issues #182: hey I am new in ODE scene and I want to know if this games will work

ps1lover, 2022-01-02:
(hey I am new in GitHub so Sorry for my bad writing)
.
.
.
  when I was child I always play my ps1 games on ps2 games like Pepsiman and crash bandicoot etc...
but the years come and Emulators have high high compatibility and the same with fat ps2 but not every game work correctly games like Monkey Hero (Formula One 99 2000 2001 by Studio 33 and sony ) they never work good on Emulators

  ( even the best Mednafen can't run Monkey Hero https://github.com/OpenEmu/Mednafen-Core/blob/master/mednafen/psx/notes/PSX-TODO 
https://www.ngemu.com/threads/monkey-hero-broken-on-all-emulators.206416/) 

 ( ps2 fat or slim is the same thing https://en.wikipedia.org/wiki/List_of_PlayStation_games_incompatible_with_PlayStation_2)

( duckstation is the same thing  but hey Monkey Hero work and  Formula games working good but in my setup they have bad sound https://docs.google.com/spreadsheets/d/1H66MxViRjjE5f8hOl5RQmF5woS1murio2dsLn14kEqo/edit#gid=0 )

and in 2020 when I heard there is ODE for the ps1 and have 100% compatibility my mind was blown but I go to the GitHub  page and I see some game not working??? like this games  (https://github.com/x-station/xstation-issues/issues/164  https://github.com/x-station/xstation-issues/issues/145 https://github.com/x-station/xstation-issues/issues/131https://github.com/x-station/xstation-issues/issues/129  https://github.com/x-station/xstation-issues/issues/128 ) so I wondering if Monkey Hero and Formula games  will work on xstation??? and thanks

ps1lover, 2022-01-02:
I love to any one to test these games and tell me if they work or not

ps1lover, 2022-01-02:
he its not close it just I am new in GitHub  and I was not seen any tutorial  so I am sorry if I do any  misunderstand but its not Closed

ramapcsx2, 2022-01-03:
Hey, all games are working. It's just that sometimes, a bug creeps in in development that can break a game. I release an update and they'll be fixed.
Right now, we're at 100% again. Your games will work :)

ps1lover, 2022-01-03:
aha so Monkey Hero and  Formula One they still not working??? ok I will wait and big thanks

ramapcsx2, 2022-01-03:
It is a low level ODE, which practically guarantees compatibility. There are corner cases that can make it fail anyway, but by now I think I've found and fixed them all :)

ps1lover, 2022-01-03:
> It is a low level ODE, which practically guarantees compatibility. There are corner cases that can make it fail anyway, but by now I think I've found and fixed them all :)

good but I never seen any video show and proof that this games working on ODE  so Its better to keep it open issue so that people can test it and  see If there is any bugs with ODE because  you have Formula  games this games  have anti piracy so yeah  thanks for the response .

ramapcsx2, 2022-01-03:
Sigh, sure :p 
I'll try and remember to test them.

ramapcsx2, 2022-01-03:
I'll close the other 2 issues, and update you here once I get to it.

ps1lover, 2022-01-04:
ok

jarnalyrkar, 2022-01-29:
I can confirm that Monkey Hero boots fine.
[image]

ramapcsx2, 2022-01-29:
Oh sorry, I totally forgot! But yea, all games work full stop. :)

ps1lover, 2022-01-31:
this is big update thanks  [jarnalyrkar  ](https://github.com/jarnalyrkar)!!! but what about  Formula games ??? they have some bad bad anti piracy and ( Maybe it's worse then spyro the dragon ???) and I thing they never have fan patch to fix anti piracy 

so your can a ask how bad it is ??? will you see if the game know you have crack cd not original cd the game will solw your cars like heavy slow and let's say you continue the game continue continue ... oh sudden !!!!   the game will stop playing and this screen will show up https://www.youtube.com/watch?v=VzcYQ1bJJvQ and after that the game will not play whatsoever.
 so yeah I wanna see if this game will ever play in ODE bruh

ps1lover, 2022-01-31:
> I can confirm that Monkey Hero works fine. [image]

but is there any bad sound or any issues ??? or is it play perfectly ??

ps1lover, 2022-01-31:
maybe this help with formula games ??? https://tcrf.net/Formula_1_Championship_Edition    https://tcrf.net/Formula_1_%2798

ps1lover, 2022-02-02:
???

jarnalyrkar, 2022-02-02:
@ps1lover I do not own those games.

jarnalyrkar, 2022-02-02:
> > I can confirm that Monkey Hero works fine. [image]
> 
> but is there any bad sound or any issues ??? or is it play perfectly ??

No troubles at all with Monkey Hero.

ramapcsx2, 2022-02-03:
F1 97 also working, tested PAL and NTSC both

jarnalyrkar, 2022-02-03:
F1 2001 working, tested PAL. Three completed laps with no anti-piracy warning.

[image]

ps1lover, 2022-02-03:
boys your legends I will close the issue because this is the only games I want to see if they work or not  and thank god they absolutely work thanks https://github.com/jarnalyrkar    and thanks for the creator https://github.com/ramapcsx2   thanks for everybody and I will see you when I buy x-station ( or better yet buy me a ode because I don't have money :) .

closed 2022-02-03 by ps1lover