x-station/xstation-issues #185: Powerslave sometimes autopauses

badass-boss, 2022-01-07:
When I play Powerslave (USA), the game sometimes auto-pauses and it's rather consistent. I'm using Hori Fighting Commander 2way and it auto pauses every once in a while even when the controller is laying unused. It is rather consistent and it auto-pauses every 3-5 minutes or so. I have 2 Hori controllers and both do the exact same thing. I tried to plug an original PS1 dualshock and so far the problem did not reoccur. However when I use Hori controllers in any other game, including fighting games, these games never auto pause and I beat so many titles and I've never noticed auto-pausing even once ever. So what gives??

ramapcsx2, 2022-01-07:
Good question.. Are you sure you're asking in the correct place? This is about the ODE ;p

badass-boss, 2022-01-07:
I have no idea. There is probably some software, hardware, xstation or incompatibility issue and it looks like something that's very complicated. This is probably the best place to get the answer

ramapcsx2, 2022-01-07:
Oh, alright.
Well, it would be a software problem, which could happen from xloader's fastboot feature. If you select full boot or restart the console to the game directly, that would then fix it.
Otherwise, the PSX library is known to have a bug here or there ;p

badass-boss, 2022-01-07:
Ah granted, I forgot about fullload. I fullloaded the game and, as expected, it still auto-pauses randomly on Hori Fighting Commander controllers. Does this indicate an in game specific bug?? I mean the solution is to just use an original PS1 controller for this and I can live with it but the question is whether the same thing will happen again

ramapcsx2, 2022-01-08:
It's a process of elimination. If you have no other thing attached and running on the PSX, except the game and the controller, then it must be a software issue.

badass-boss, 2022-01-14:
I think the issue may be completely unrelated to xstation. I noticed those Hori controllers don't have many of the resistors completely while original Sony controllers are loaded with resistors and some semiconductors. Maybe there is some ripple effect going on. If I manage to solve the issue I will provide the solution regardless

ramapcsx2, 2022-01-15:
Sure, I'll leave the issue open for a while longer. Good luck :)

badass-boss, 2022-02-07:
I had no luck in adding resistors and so on. Same exact issues but seriously though, I don't think it's a problem with Xstation. I also checked the original PS1 controller without analogs and the game works just fine. I consider this issue "solved" just by using original Zony controllers. Hori controllers are out of luck only for this game for some reason but still, everyone is more likely to just have original controllers and this is what matters. Why is that the case just for this specific game is a mystery

closed 2022-02-07 by ramapcsx2