x-station/xstation-issues #186: Myst- Japan Original and Rev 1 version has graphical issues when changing between scenes

VirtuaCop, 2022-01-09:
The two japanese versions of Myst below:
http://redump.org/disc/4786/
http://redump.org/disc/33887/
When transitioning scenes left and right the graphics will become corrupted and flicker. This happens both with fast load and full load.

I don't have a way to test with a real disc or burned disc but on Xebra Emulator with the same disc image the issue was not seen.

The issue also does not happen on xstation with the Japanese Rev2 image or the US image
http://redump.org/disc/1488/
http://redump.org/disc/11129/

Since this is such a specific bug to only the first two revisions of the japanese version, I would consider it low priority. But maybe it can lead to better future compatibility. Who knows. Thanks.

ramapcsx2, 2022-01-09:
Thanks for reporting, I'll have a look :)

ramapcsx2, 2022-01-09:
I get this glitch as well. It doesn't seem to react to ODE timings at all, so my guess is that it's a bug in the early releases.
But I'll check it out on unmodified consoles. Let's see what happens :p

ramapcsx2, 2022-01-09:
It's a buggy game, sorry. Real drive console:
https://i.imgur.com/mu988ka.jpg

VirtuaCop, 2022-01-09:
Cool thanks for checking. I didn't have another way to read a disc on a real hardware.

closed 2022-01-09 by VirtuaCop