x-station/xstation-issues #187: XStation not booting on full boot, only reset

rwebster85, 2022-01-10:
Hi mate. Only started having this issue today. On a power on boot it hangs on the white logo, then after several seconds the default PS menu appears (memory card manager, music player).

However, when preset Reset the console gets passed the boot logos and into XStation.

This was on the latest firmware but also tried downgrading to 1.4.4.

I have a PU-18 5552 model unit, with the X-Reset mod also installed. Resetting the console using X-Reset button combination also boots into XStation fine.

It's just from a cold power-on boot that it's not working. Any ideas?

rwebster85, 2022-01-10:
> Hey, the QSB might have a cold solder joint. If you can, try reflowing all the solder points.

Happy to give that a try mate. Be a few weeks before I can get to it though. Will report back.

rwebster85, 2022-01-10:
Tried a different SD, as well as without using the SD adapter. Same behaviour. Strange. About 1 in 10 times it will boot from power on.

I'll look at the solder in a few weeks.

rwebster85, 2022-01-12:
I managed to get chance to look at this today.

TL;DR: reflowing worked 😁 

Before doing anything I decided to check continuity. Everything seemed to be fine. So I started reflowing different areas at a time.

Eventually got round to this area:

[image]

Reflowing the solder points here resolved the issue. Not sure if it was one of them in particular though.

ramapcsx2, 2022-01-12:
Glad to hear :)
I'll close it and label it for reference.

closed 2022-01-12 by ramapcsx2