x-station/xstation-issues #188: Late PU-8 install.  PS1 detects video, but shows black screen.

Orcut, 2022-01-11:
Hello everyone, 

I recently did 2 installs on late PU-8 consoles.  Before the install, the consoles would both boot the white screen PS1 logo, the, go to the memory card screen.  After I did both installs, both consoles will detect video, but remain on a black screen. No white boot screen at all.  I have verified the QSB with a multimeter and everything seems to be in order there.  Instructions were followed and all necessary resistors were removed.  Does anyone know what could be causing my black screen issue?

ramapcsx2, 2022-01-11:
Hey,
that sounds like the APLL wire isn't connected, or is shorting out. It might be touching the case metal.

Orcut, 2022-01-11:
[image]
[image]

[image]

For reference, I am not testing it in the case, so there is no metal the wire could be touching.  Pictures for reference.  Not sure what else to check.

ramapcsx2, 2022-01-11:
Is the xStation blue LED on? Any activity on the green LED?

Orcut, 2022-01-11:
When the console is powered on, the LED is a solid blue.  No activity on the green LED.

ramapcsx2, 2022-01-11:
Check the green LED carefully. It should at least briefly blink up on power up.
If it never does anything on a power cycle, the ESP32 on it isn't running code, ie it's been fried.

ramapcsx2, 2022-01-11:
If you can, please measure the voltage that goes into the regulator on the xStation. 
I'm interested in what comes in, and also what comes out of it. (Should be around 8V in, 3.3V out)

Orcut, 2022-01-11:
[image]
[image]

I think I did it right?  Not sure though.

ramapcsx2, 2022-01-11:
Not quite right (black probe can go to any ground, the ESP32 module itself is good), but I can see that the voltages are good.
If you get absolutely no action on the green LED when you power it up, and it also never goes to a steady on/off blink when running, then the xStation needs to be repaired.
In that case, the ESP32 is simply not working at all.

Orcut, 2022-01-11:
I think I got the voltage figured out.  It looks like it is getting 7.25v and 3.22v.  Not sure if that makes a difference.  But I am 100% certain that I am not getting any green light.

One more question.  Even with the install done to the motherboard, shouldn't I at least get the white PS1 logo screen regardless if the Xstation is plugged in or not?

ramapcsx2, 2022-01-11:
With the modification to the APLL point, the PSX board now requires the xStation to be running.
If it doesn't, then the CPU doesn't get a clock, and the screen stays black.

3.22V sounds a bit low, but that could be just the meter.

Orcut, 2022-01-11:
Thank you for all your help!  I guess the best thing to do at this point would be to contact Castlemania about the defective unit?

ramapcsx2, 2022-01-11:
Yes. It most likely is dead, sorry ><