x-station/xstation-issues #189: Bend so much or with a very sharp angle the flex cable prevents Xstation from boot correctly

miguelk5, 2022-01-12:
Hello, I've got a console from Will's Console Modifications, that a client of him sent me since it doesnt boot the Xstation menu most of the times.

Even if there is a good job in the joins I saw some details: unnecesary plastic cut, removed back plate, removed the front cooper protection.

But the worse thing I saw and the cause of the xstation malfunction is that he bended the cable very strongly and at a very sharp angle:

[image]

You can see on his Facebook's page that he bend so much the flex cables:
https://www.facebook.com/WillsConsoleModifications/posts/1575015022698362

Xstation install guide:
[image]

Risk of bend a flex cable more than minimum bend radius, or at very sharp angle:

"All flex cables have a minimum bend radius, under which a specified number of fold-unfold cycles is guaranteed. This radius depends on the brand, but generally it's around 5-10 mm. Here's a spec for AWM 20941 flex cable from JUDD WIRE, which is sold on Amazon as a Raspberry camera accessory"

"What happens if you bend it with a smaller radius is not specified. In practice, this means that the cable may break sooner, or even instantly."

Source: https://raspberrypi.stackexchange.com/questions/47685/can-the-rpi-camera-cable-be-folded-creased

So said that I knew that the source of the problem were the flex cable, not the sd card slot, so I tested the PS1 wiht a working SD card (it works on my own PS1) and the original flex cable, the Xstation menu didn't boot, I repeated the test with a new flex cable and this time the Xstation boot correctly. I have repeated the test 5 times with the same result.

I think it would be a good idea to clarify in the guide the risks of bending the cable too much and I don't know if is a good idea keep Wil Console Modifications as a recommended installer, finally the guy payed like 107 € (122 USD) : 20L customs fees , 40L service, 30L shipping, and after that the xstation is not working so he had to send the PS1 to me which cost 10€ + 10€ ship back to his country + 10€ the flex cable + 15 € for a broken ps1 to put again the backplate + 20 € my services. He had to wait 6 months since Will didn't knew how to declare the item correctly to pass the customs.

I have talked with Will on Facebook, I sent him a video of my test with his cable, and send him the picture of the flex cable from the install guide, the article which talks about the risk of bend a flex cable too much, even so he doesn't recognice that is not a good idea manipulate the flex cable like that:

"No I didn't, it worked perfectly, I have done 100+ installs and never a problem"

I want to know what do Xstation developers think about this flex cable thing, and as I have said if is good idea keep on recommended installers people who do a fault (thats shouldn't be a problem) but after that don't recognize it, don't offer a solution or a partial refund at least.

Regards.

ramapcsx2, 2022-01-13:
I'm not quite sure why you posted this here, but I'll reply my thoughts.
First of all, I agree that bending these flex cables tightly is an issue and should be avoided.
However, we don't have data on that.
It seems reasonable that if you want to bend it, then do it properly once, and never undo the bend later.
Bending back and forth would cause degradation for sure.
But bending once and leaving it? I'm not sure.
I can bring it up with Will over on Discord. I do agree that it isn't necessary.
The datasheet doesn't specify, and the one you linked to talks about bending/unbending cycles, not single bending.

ramapcsx2, 2022-01-13:
Okay, I have the full picture now. I think you should settle your qualms with Will and move on.

miguelk5, 2022-01-14:
In my opinion there is not full picture, a guy spent 122 USD for nothing, I think that the guide could emphasize more on the risks when fold the flex cable.

L10N37, 2022-02-04:
This happened to a system on mine without a fold there, I just confirmed by laying the system on it's side and jiggling the cable. At first i managed to get the system actually firing up, had to jiggle it around more to get it actually go the black screen and proceed to Xstation OSD