x-station/xstation-issues #190: Late PU-8 23 revision

ZweiGP01, 2022-01-19:
Ok, bout to beat a dead horse here but here goes. Checked and triple checked joints to the test pins and to the pin header on the ffc connector. not getting anywhere. Please Help! thanks. also using a RePSX PSU doubtful that has anything to do with it.
[image]
[image]
[image]
[image]
.

ZweiGP01, 2022-01-19:
Also forgot to mention that it does write a config file but just sits on the Memcard/CDplayer menu.

ramapcsx2, 2022-01-19:
Could be many things, among them chip damage. Did the console work before?
The board revision itself is supported.

ZweiGP01, 2022-01-19:
Yeah, just had a bad laser when i bought it. i had another laser lying around so swapped it and it spun up. im gonna have to check all the chips. ill get back to you when throughly look through em in a bit.

ramapcsx2, 2022-01-19:
Just to make sure: The lid switch was closed, right? :p

ZweiGP01, 2022-01-19:
> Just to make sure: The lid switch was closed, right? :p

God at this point i would prefer for it to be that. i actually checked the switch and checked the traces for it. even went as far as to bridging it to see if it was the problem my ocd demanded i also remove it clean it and put it back on.

ramapcsx2, 2022-01-19:
If you have an expansion port module, put Unirom onto it and see whether it can boot games.
Example: https://i.imgur.com/BcJfti8.jpeg
Unirom: https://github.com/JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1

ZweiGP01, 2022-01-20:
i ordered one last night, once it gets here ill let it rip.

brendanpoulsbo, 2022-01-22:
Hi @ZweiGP01 can you let me know where you ordered yours? I also need to test with one but having a hard time finding a place to order

L10N37, 2022-01-28:
UniRom is working on memory card ports with a memory card, use memory card annihilator on a modded PS2, else something like memcarduino, else a chipped playstation of the same model to use the unirom boot disc to create it.

Matt90116, 2022-02-04:
This is off topic, but I am slightly concerned snd didnt know where to post this I am brand new to github. I was lifting pins on IC701 and when I got to 50's I managed to lift on IC308 not noticing. I went ahead and pushed them back and reflowed not installing the xstation yet and I tested it to make sure all was correct. If the xstation isnt installed after this point will you still get sound? The boot menu takes 10 seconds now and I dont get sound anymore. I hope once I install the xstation that will change because I reflowed those 3 pins back nicely, but no difference from before separating all the pins when I had sound

miguelk5, 2022-03-18:
> If you have an expansion port module, put Unirom onto it and see whether it can boot games. Example: https://i.imgur.com/BcJfti8.jpeg Unirom: https://github.com/JonathanDotCel/unirom8_bootdisc_and_firmware_for_ps1

I have a similar issue with a NTSC JAP PU-8 I have unirom, but do I need to undo the Xstation install to run Unirom right ? 

https://github.com/x-station/xstation-issues/issues/202

VashTS000, 2024-07-30:
I have the exact same issue, spent 5 hours today troubleshooting. All the solder points are good, tested from component to xstation ribbon. I can get into unirom but it freezes on loading PSX.exe, on the very first time I loaded unirom it said loading xstation.exe 

Really frustrating to have issues, I would appreciate any help you can provide!
[image]

VashTS000, 2024-07-30:
Couple other points - loading CD player does work and it "plays" the CD. Booting into unirom again now it does say xstation.exe I changed nothing, just wanted to give it another go before bed. 
[image]

VashTS000, 2024-07-30:
Sorry for spamming but removing 2.0.2 and adding 1.6.1 gives a different message in freePSXboot 
[image]

ramapcsx2, 2024-07-30:
Hey, yours appears to be a soldering issue.
The fact it can say "xstation.exe" means everything is fundamentally working, but then the problem prevents it working fully.
(The .exe name is data from the virtual disk, which can only show up if everything was working at least for a little while.)

I suggest reflowing the QSB, remove and reattach the flat flex cable, then try again.

VashTS000, 2024-07-30:
I have continuity from the board to the xstation itself on all points. What more could I need to make sure there is a solid connection?? I could wire it manually and take out the qsb
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

VashTS000, 2024-07-31:
It looks wet from isopropyl alcohol 😁 I removed the qsb and I'm going to wire from the board to xstation directly. I might go to the qsb fpc connector and use the ribbon, but I'm worried that the ribbon could an issue causing intermittent connection.

I'll prob go from board to xstation directly to avoid any other possible issues. Wish me luck!

ramapcsx2, 2024-07-31:
Alright, sure :p
There probably was no need to remove the QSB, but if you want to be able to ensure each single connection, then go ahead with direct wiring.
Make sure you use a bit thicker wire for 8V and main GND, all the other signals can be very thin.

ramapcsx2, 2024-07-31:
Also keep in mind that the QSB itself grounds some points on the PSX board.

VashTS000, 2024-07-31:
Well there are 3 ground pins, is that enough for the xstation to operate?

ramapcsx2, 2024-07-31:
The QSB has 1 point to ground itself, but distributes that ground to other locations. You'll be able to measure where those are with the QSB removed.

VashTS000, 2024-08-01:
I decided it's too much work to complete the wiring and $30 gets me a 5501 so I'm taking that route. Should be here Monday, hopefully I get up and running that day.

ramapcsx2, 2024-08-02:
Oki, remember to press down on the QSB a bit when soldering the first few spots.
You want the QSB flush to the mainboard, making problems much more unlikely.

VashTS000, 2024-08-03:
5501 board is in, and it's done, all working fine. The big question - what was wrong with the late pu8? I strongly would suggest my soldering was not to blame.  I've been doing rgh3 on Xbox 360 for like 4 years and I've done some OLED and lite hwfly installs. 

Can it be software related and how would we find out?

ramapcsx2, 2024-08-03:
Good that it works in the other machine :)

Debugging the old one is a bit of an art, really. I suspect either a soldering issue, or chip damage on the PSX (decoder, SPU, maybe). But it's unlikely to have anything to do with any software.

What you can do now depends on what tools you have.
Ideally there is a scope, with which you can probe around the board. I've had PU-8 where the SPU would produce signals like this:
https://imgur.com/09S9Tts
Clearly bad IC, and this would require a swap with a working part.

VashTS000, 2024-08-04:
Dang I don't have an oscilloscope (or much knowledge on it!) but it's just hard to believe that it was the PS1 at fault. Maybe the earlier boards are more likely to have hardware issues than the later 5500 series 😬

Anyway, I'm glad it's working too! Thanks for your attention on this!!

ramapcsx2, 2024-08-04:
PU-8/7 do have this more often, yea, but it's just one of the possibilities. It really depends on your tools and what you already know how to test... Without scope, most people resort to careful multimeter testing and simply trying to reflow everything.