x-station/xstation-issues #191: Two PU-18 installs, both have same problem. Were working fine at first but now won't go past memory card/cd player screen. One green light blink on powerup.

brendanpoulsbo, 2022-01-22:
Hi all!

I was happy to test my newest installs and find that they worked, but after doing some "stress testing" (aka power on and off a hundred times) I found that they were initially inconsistent. They would work 90% the time powering on, then about 50%, now they don't work at all. They just get to the PS menu for memory cards/cd player.

Here are some pictures of my installation in the second console: https://imgur.com/a/fJY8nyv

I have done the following:

1. Made clean joints and have a good alignment with the QSB.
2. Reflowed the pins of the IC chip and I ensured they are not bridging.
3. Checked continuity with my multimeter on both systems. Using the diagram in this FAQ for PU-18 boards, they retain continuity on all test points.
4. Also checked continuity on all pins of both of my FFC cables, and they retain continuity as well. At first I thought that was the failure point because of the initial inconsistency. It seemed to work better when the console shell was pushed down by my hand, or the screws were tightened manually. This is no longer the case - doesn't work at all.
5. Tried multiple microSD cards as well as the sd card extender. This does not seem to be a failure point, as I have never experienced the consistent green flash that indicates sd read failure. The extender was used when the consoles first worked - I do not think this is the issue. 

Now two days after the initial successful install they just don't boot into the Xstation menu. 

Here's the lights on the Xstation. Happening on both consoles. At first when it was working, I remember green flashing twice on power on. Now green does once instead of a quick two, then moments later does a really fast flash https://i.imgur.com/b0vXS0p.gifv . I do not know what this means, but it's the same across both consoles. I really hope this might be a clue of where to check. 

Thanks so much for any assistance or pointers. I was so happy when these installations worked at first, now I'm super sad they're not :( I will try anything.

ramapcsx2, 2022-01-22:
Hey,
I think it is a signal issue, so some signal not making it to the console.
The LED pattern is once for the initialization and mounting (probably the loader), then after the Mechacon reset, it's loading sectors continuously.
The system must somehow not be able to work with the data, likely a signal isn't working.

As with the other open issue currently with a similar problem: Do you have an expansiom ROM cart to run Unirom?

brendanpoulsbo, 2022-01-22:
Hi rama thanks for the quick response!

Do you think the FFC could be damaged resulting in a degraded signal? Could this cause the light pattern I see on the Xstation units? Possibly do you know of a way I can test the quality of the signal with a multimeter? I ask because although I initially bent them very lightly, after many times of removing and replacing all the parts, I believe the bend has become harsher. 

I do not have an expansion ROM cart but I will purchase one right now and update when I can test it. 

Thanks again!

ramapcsx2, 2022-01-23:
You can't test these signals with a multimeter, no.
I'd say the flat flex is fine, unless it looks really bad.

Any other method to run homebrew would work. There's something for Memory Cards, right?
You just need to be able to run Unirom on the xStation console. And it'll just give you an idea which signal might be bad.
If it was the license stuff, Unirom would even boot the game / loader.

L10N37, 2022-01-23:
I have 2  returns with this issue to fix this coming week. both PU18. my fully recapped personal console did this too (both PSU and mainboard).

the Xstation itself wasn't issue as I transplanted into another system for a customer when I ran out of Xstations along with full dual region and IGR... had perfect feedback and no issues since.. been at least a couple months since that.

one of the returns I'm happy to conclude it was just bad luck, these systems are over 20 years old... something on it failed. BIOS audio cut out (no start up sounds) 

I actually transplanted that into yet another board and that one gets to a black screen and then shit itself... 

I have enough systems to transplant both installs to new mainboards (one will be on its 3rd board lol) 

Im pretty satisfied to call it a run of bad luck and look no further Into it. I'll just add the boards to my graveyard... wait... more of a soul pool.. awaiting possible revival at a very future date.

a console that sells this many million units isn't too big of a concern that you'd have a cry over some dead mainboards :P 

customers paid good money so happy to do what's best for them and I'll grab another batch of xstations soon

L10N37, 2022-01-27:
Finally checked these boards, of course i was going to do a little check over before just swapping them. Check PS602, both these had no BIOS audio and wouldn't launch Xstation. Both blown. Looks to be a 1a fuse. So 3 returns last week and all blown fuses (got a PU22 that stopped spinning discs too)

brendanpoulsbo, 2022-01-27:
Okay I have a really big update so bear with me. I have made some progress and have findings that are common across both installs as well as new issues that are specific to each. Here goes:

Common:
I got both Xstations to work, and to work consistently! However, they BOTH stopped working completely, again. I happened to get them working by reflowing some solder points as well as the pins on the IC701 chip. Both of them seemed to fail after leaving them on for some time, seemingly after the PSU warmed up a good amount. However I read in another issue that temps should not be an issue for the Xstation hardware, so I hesitate to call this the issue. I swapped both PSUs during testing while one was working and the other wasn't - did not change either of the units' functionality. This goes against the common sentiment of cold solder joints - where the functionality gets better as they warm up. It seems so far in my case, it gets worse.

Install 1:
At first, install n1 was working after I cleaned up flux residue on the IC chip and reflowed all its pins on the sides that were lifted. I did a quick test of about a dozen power on/off/resets with no issues loading the Xstation menu. Then, as a bit of a stress, I decided to leave it running on the Xstation menu for about 30 minutes while I worked on the other. Came back to it and selected a game, and it would not load. I turned off the console and back on - now it hangs on "Please insert Playstation CD Rom". (Yes, I reflowed pin 3 [#181] and checked all the pins on the mechacon, didn't change anything). Currently this unit does not load the PS Mem Card screen or Xstation menu, it's always the CD ROM page.

Install 2:
At first, install n2 was very inconsistent, it would work about 50% of the time without a good pattern of failure or success. However the specific error was very consistent - it would always load into the Playstation logo / SCEA screen, but FAIL to load the Xstation menu - it would just hang. After reflowing the IC pins and cleaning up more parts of the install, I believe it just started to load fine. It got to probably the most consistent loads of anything so far, so I put it through a stress test in the menu like n1. It stayed at the menu just fine for longer periods. Then I loaded up MGS and let it stay at the intro video loop. It also worked great for a while, but unfortunately after a test of about an hour I came back to see the spinning disc logo. Turned off the console and now it never gets past the Mem Card / CD Player screen. Since that MGS test I haven't been able to get it to load the Xstation menu.

Further troubleshooting taken:
1. I bought new liquid flux as my old flux seemed to be leaving a lot of residue and not cleaning properly. After reflowing the relevant pins (3 for mechacon and 4 for possible bridging with gnd) it didn't have any effect on loading.
2. I tested continuity again on each board, but this time checked from the pins on the QSB FFC connector to the test points. All had continuity. Again, my flat flex cables have full continuity.
3. I bought a Unirom Gameshark for the expansion port. I don't know how to actually operate this, but a quick search resulted that it should just load up when plugged in. This is NOT the case for either of my units. They exhibit the same behavior as if nothing is plugged in to the IO port at all (n1 goes to CD-ROM screen, n2 goes to Mem Card / CD Player). I realize the Gameshark could be not properly flashed or broken or something, but given that it was $35 from a reputable seller I hesitate to go with that assumption. 
4. I reformatted my SanDisk SD card using the SD Memory Card Formatter tool. It did exFat automatically, I added back the proper files in the 00xstation folder. Didn't have any effect.

So now my next steps might involve that Unirom card or the maybe the fuse listed above. @L10N37 can you possibly point me to the location of that fuse so I don't have to do more digging (I'm exhausted with these installs as you can imagine, ha). And @ramapcsx2 do you think this Unirom card could be a further lead and if it's telling you anything that it doesn't load up automatically?

L10N37, 2022-01-27:
Some notes on these systems that slowly died like above:

With the lid button down, the controllers are almost non responsive in the BIOS screen.  You have to press right continuously for about 10 seconds or more to have it move across to the AUDIO player and vice versa, same goes with getting into the memory card manager. Unirom will not be launch off FREEPSXBOOT with the drive lid pressed.

With the drivelid not pressed, no issues like above. I can launch unirom (using 'i' instead of latest 'j' version, due to J only being on a boot disc at this stage and not having a chipped PU18 on hand to create the card, thus using memcarduino/memcard rex)

STATUS in unirom reports "UNKNOWN EEPROM! (FF,FF)
CRC32: D786F0B9 (V3.0 01/06/97 E)

Attempting to launch the Xstation as a CD (closing the drive lid button and launching CD) results in the solid blue light staying blue, and the green light blinking for a shortwhile in what I assume is an attempt to launch the Xstation loader.

I will update this if i find anything else

L10N37, 2022-01-27:
> Okay I have a really big update so bear with me. I have made some progress and have findings that are common across both installs as well as new issues that are specific to each. Here goes:
>
> Common: [...]

> Okay I have a really big update so bear with me. I have made some progress and have findings that are common across both installs as well as new issues that are specific to each. Here goes:
>
> Common: [...]

Sorry this just came through now!
The fuse should just be for the rumble pack! not sure why both of these had no continuity...
I am in the process now of switching out IC's from boards to try and ascertain what's failed as well...!  Fair enough to just give these customers new mainboards altogether (which, i will actually be doing) but I really want to know what's gone wrong.

One was sold in October and the other didn't even last that long. I've sold over 10 of these, it's not some definite impending failure (as far as I can tell) but a bit suspect with my 2 returns, my dying personal console (started loading to the loader less and less successfully as time went on. this was a fully recapped system) + your 2 PU18's on top it!

L10N37, 2022-01-27:
Look if it's a failing CXD2545Q you can get these out of old sony CD players or what looks to be NOS parts available online.
I'm not willing to transplant one from a perfectly working PU18 onto a non working for obvious reasons... I have 3 X stations to prepare today and now it's 7pm ..oh god. I better get into it, will check up here at a later date :)

ramapcsx2, 2022-01-27:
@L10N37 your issues are consistent with "half bad" 1815Q FIFO chips. They will kind of decode data, but not always / not good enough to work.

@brendanpoulsbo Your consoles sound like still a cold solder joint but working chips. Heating up doesn't have to improve or worsen the situation, it just changes it. That kind of behaviour is indicative of minimal movement opening the cold joint.

L10N37, 2022-01-27:
> @L10N37 your issues are consistent with "half bad" 1815Q FIFO chips. They will kind of decode data, but not always / not good enough to work.
>
> @brendanpoulsbo Your consoles sound like still a cold [...]

Thanks Rama,

These boards will just be put aside for warm ups / spares.
Just did a fresh install and thought it might be worth noting that playing around on the dud boards earlier with Unirom did seem to corrupt one of the cards 'loader.bin' files, 4.8Mb file and it was there as 'zero bytes'. 

Other than that, all good. None of the IC's to be heating up or anything. Probably a streak of bad luck. Managed to find 4 more PU18's in the pile so i'll still have 1 extra for the next batch :)

ramapcsx2, 2022-01-27:
Very curious. I've never managed to corrupt a file on the sd card, and I tried to deliberately :p

brendanpoulsbo, 2022-01-28:
Thanks guys for helping me as I lose my sanity over these boards! I'm getting closer and closer to just ordering new consoles as this is ridiculous.

One day later, I got both of them to work and then stop working, again. I reflowed all the joints on both boards as well as the pins on the IC chips. Checked continuity from FFC pins all the way to test points. Everything checks out.

I took new pictures to show you guys the current state of things. Note how on n1 I added a wire on the bottom left ground point. This is because while it initially passed the beep test, after reflowing it wasn't making a good joint, so I wired directly to the pin. Now it's passing from test point to FFC (1,11,18) as well.

Updated N1: https://imgur.com/a/tsQ8Sqd 

Updated N2: https://imgur.com/a/BnokcXE 

Any more ideas would be great, otherwise I'm getting new boards. As of writing N1 is working again, but I'm expecting it to die on MGS intro within the hour. Thanks guys.

ramapcsx2, 2022-01-28:
New boards will give you a new attempt to get it right, nothing more :p 
I think that, since the current boards sometimes work for a while, that there is a hidden fault that develops with temperature cycles. This is clearly very hard to debug, so it's up to you how long you want to try.
Gotta say though, N1 looks horribly corroded in general.

brendanpoulsbo, 2022-01-28:
Oh really? Where does the corrosion look bad to you? 

I'm trying to clean them up extra good and add nothing unnecessary as I continue work on them. I was worried about corrosion going forward, but I thought that I had cleaned it up to the best I can at this point. (Been doing a lot of scraping on microscopic areas to get any residue and corrosion off). Tips on cleaning/protecting? ;)

That being said N1 is still surprisingly running the intro movie after an hour. N2 has been giving me more trouble over this whole process.

L10N37, 2022-01-28:
your issue sounds extremely intermittent and like a connection just isn't quite right.

one of mine did have an intermittent issue prior to dispatch but I thought I'd debugged that one. It basically had issues prior to the install. The other, ehhhh... not sure.. but honestly it was a $5 op shop pick up (I generally pay A LOT more :P )

as for my personal console, I have a feeling it was the drive lid button playing up as that was an issue initially being sticky. (I chose the rustiest, piece of crap system system I could find that looked like it lived in a backyard, had capacitors literally falling off the PSU, then recapped the entire system and gave it a new life) 

anyway, that's probably why It didn't drive me crazy, as these systems were dead on return. not intermittent. 

I re-used a QSB on a return (I was open with customer about the bodge)

https://imgur.com/a/LmcrANi

I found those points broken
on the other there were other points too. the ones on the outer perimeter are SUPER sensitive. may I suggest ,as a last resort ...bridging the outer joints across to the cable header ??

brendanpoulsbo, 2022-01-29:
Welp, I'm not going to jump the gun and say they are fixed, but I was able to get them both working again just by deep cleaning especially between the pins of the IC chip and the mechacon. They both survived the MGS test for 3+ hours each, and more importantly, were still working the following morning when I booted them up!

I did add 4 wires on the edge points like you suggested, but I think the cleaning is what made n2 work as at first it still froze on the SCEA screen with the wires. 

Gonna keep testing them for the next several days, probably weeks, lol. Thanks again for the help guys.

closed 2022-01-30 by brendanpoulsbo