x-station/xstation-issues #193: Potential memory card dumper

badass-boss, 2022-02-06:
So here's a functionality idea- is this possible to implement a memory card dump functionality so the MC contents can be copied to the SD card and restored back from SD to MC via xstation?? Fenrir for Saturn has a similar functionality and if this gets implemented then that functionality would be PURE GOLD! We all know memory cards for PS1 suffer from insufficient space available all the time and they get full all the time. This sort of functionality would require an average user to just use 1 memory card (or maybe 2 if you want to play RPG Maker) and save infinite amount of times for all games available. Please consider this!!

ramapcsx2, 2022-02-06:
It's actually already coded, but not integrated yet :)

badass-boss, 2022-02-07:
! ! ! ! !

ramapcsx2, 2022-05-15:
You can find a bios dumper tool on CD, and use that :)
I think it was one of Shendo's tools.

xenyou, 2022-05-15:
do you mean this?
http://shendosoft.blogspot.com/2013/07/psx-bios-dumper-26-released.html

I already tried this image with xstation but it does not boot.
And I found this log in issues.
https://github.com/x-station/xstation-issues/issues/159

ramapcsx2, 2022-05-15:
Hm, try fixing the image with https://www.romhacking.net/utilities/1550/ ?

xenyou, 2022-05-16:
I tried to fix the image with that tool.
But the corrected image also does not boot. Hmm..

ramapcsx2, 2022-05-16:
Yeah, the release is totally miss-mastered. It's not repairable in real time on xStation.
I re-mastered it here:
[biosdumper_fixed.zip](https://github.com/x-station/xstation-issues/files/8701747/biosdumper_fixed.zip)

xenyou, 2022-05-17:
It works now!!
Great thanks to @ramapcsx2 .

badass-boss, 2022-06-18:
OK so maybe there exists an ISO image that can dump memory cards as well?? Am I missing something here?

ramapcsx2, 2022-06-20:
The topic briefly changed to dumping the console BIOS.
XStation at this point can copy your Memory Card saves from one card to another, same as the BIOS function.
It is planned to add a function to dump a card and restore to / from the xStation SD card.

Z3R0XZ, 2022-09-06:
Oh, yes with this feature Xstation will be perfect!

Xenos-Interactive, 2026-04-01:
> It's actually already coded, but not integrated yet :)

So is there any news, if this has been implemented? Or do i need a third device like the MemCard Pro to dump the saves?