x-station/xstation-issues #194: Black screen and image stuttering

Sklzz, 2022-02-09:
Installed xstation on scph-5502.  All points are soldered correctly, there are no short circuits, all connections are in order from beginning to end.  Used several sd cards, no change...
  When you turn on the console on the xstation board, a green signal appears 2 times.  The blue light is on constantly.  If you open the drive cover, the console menu is loaded.  With the lid closed, a black screen with twitches resembling a resolution change ... Please help me figure it out

Sklzz, 2022-02-09:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ramapcsx2, 2022-02-09:
It's obviously a major malfunction that prevents the CPU from starting up when the lid is closed.
The installation pictures tell me that you should send it to be professionally repaired :)

Sklzz, 2022-02-09:
> Очевидно, это серьезная неисправность, которая не позволяет процессору запускаться при закрытой крышке. Фотографии установки говорят мне, что вы должны отправить его на профессиональный ремонт :)

А что не так с установкой? Я QSB уже 2 раза перепаивал, полностью.

> It's obviously a major malfunction that prevents the CPU from starting up when the lid is closed. The installation pictures tell me that you should send it to be professionally repaired :)

What's wrong with the installation?  I already soldered QSB 2 times, completely...
![IMG_20220209_175524](https://user-
[image]
[image]
[image]
images.githubusercontent.com/99308125/153227141-eb29a1fa-5ffc-4417-9b43-e6051d61f829.jpg)
[image]

Sklzz, 2022-02-09:
[image]
[image]
[image]
[image]
[image]

Sklzz, 2022-02-09:
Самое первое включение было нормальным, меню xstation загрузилось, но белый цвет изображения стал розовым, потом я отключил питание, и после повторного включения появился чёрный экран...

ramapcsx2, 2022-02-09:
That's the problem: I can't debug it from pictures. I only know from your error description that there is something like a short developing. Seeing the pictures, I can only recommend you send it to someone else to repair it.

ramapcsx2, 2022-02-09:
"The very first power on was fine, the xstation menu loaded but the white color of the image turned pink, then I turned off the power and after turning it on again a black screen appeared..."
Yes, a shortcut somewhere, most likely.

Sklzz, 2022-02-09:
> "The very first power on was fine, the xstation menu loaded but the white color of the image turned pink, then I turned off the power and after turning it on again a black screen appeared..." Yes, a shortcut somewhere, most likely.

https://youtu.be/X9ivN5zbkBU

ramapcsx2, 2022-02-09:
Yes, the shortcut has to do with the lid switch.
More importantly, you almost touched the live power supply! Be careful!

Sklzz, 2022-02-09:
> Yes, the shortcut has to do with the lid switch. More importantly, you almost touched the live power supply! Be careful!

Демонтировал плату QSB ещё раз, заметил под ней очень много флюса, как этого избежать?

Sklzz, 2022-02-09:
> Переключатель крышки — это вход на ESP32, и только выборочный. Это приводит либо к тому, что ESP32 начинает отправлять данные и сигналы, либо не отправляет данные и сигналы. Поскольку вы получаете черный экран, когда переключатель крышки закрыт, ярлык становится критическим, когда ESP32 пытается поднять или опустить линию. Какая это линия, я не могу сказать. Все что на фото похоже придется переделывать. Используемые провода слишком толстые, места пайки выглядят как перемычки.

Sklzz, 2022-02-09:
> > Да, ярлык связанный с переключателем крышки. Что еще более важно, вы почти коснулись источника питания! Будь осторожен!
>  Нашёл короткое замыкание контакта 8 и GND, откуда оно могло взяться?
> > Переключатель крышки — это вход на ESP32, и только выборочный. Это либо приводит к тому, что ESP32 начинает отправлять данные и сигналы, либо не отправляет данные и сигналы. выявляется высокий уровень черного экрана, когда переключатель крышки закрывается, вероятность становится доступной, когда ESP32 пытается поднять или опустить линию. Какая это линия, я не могу сказать. Все что на фото возможно переделывать. Используемые провода слишком толстые, места пайки выглядят как перемычки.
> 
> Спаял снова, чисто, без следов флюса, проблема осталась

Sklzz, 2022-02-09:
> The lid switch is an input on the ESP32, and only sampled. It has the effect of either the ESP32 going to start sending data and signals, or not sending data and signals. Since you get a black screen when the lid switch is closed, the shortcut is becoming critical when the ESP32 tries to drive a line high or low. Which line it is, I can't say. Everything in the pictures looks like it will have to be redone. The wires used are too thick, solder joints look like they bridge.

Found a short circuit of pin 8 and GND, where did it come from?

Sklzz, 2022-02-09:
> > Переключатель крышки — это вход на ESP32, и только выборочный. Это приводит либо к тому, что ESP32 начинает отправлять данные и сигналы, либо не отправляет данные и сигналы. Поскольку вы получаете черный экран, когда переключатель крышки закрыт, ярлык становится критическим, когда ESP32 пытается поднять или опустить линию. Какая это линия, я не могу сказать. Все что на фото похоже придется переделывать. Используемые провода слишком толстые, места пайки выглядят как перемычки.
> 
> Обнаружено короткое замыкание контакта 8 и GND, откуда оно взялось?

Правка: забыл отпустить кнопку закрытия крышки,  из-за неё происходит замыкание 8 на gnd

Sklzz, 2022-02-09:
Какие ещё контакты можно проверить?

ramapcsx2, 2022-02-09:
I'm sorry, I can't help you further. It requires an installer with debug tools, such as an oscilloscope, and with experience looking for faults.

Sklzz, 2022-02-09:
> I'm sorry, I can't help you further. It requires an installer with debug tools, such as an oscilloscope, and with experience looking for faults.

it's a pity (

ramapcsx2, 2022-02-10:
What I can recommend is this: You still have a fresh QSB for a PU-8 console, right?
Try finding such a console, and retry the install.

Sklzz, 2022-02-14:
> Что я могу порекомендовать, так это: у вас все еще есть свежий QSB для консоли PU-8, верно? Попробуйте найти такую ​​консоль и повторите попытку установки.

I bought the SCPH-5552 console, installed and checked everything, when you turn it on, xstation still cannot start (((just like on the previous SCPH-5502...

Sklzz, 2022-02-14:
> What I can recommend is this: You still have a fresh QSB for a PU-8 console, right? Try finding such a console, and retry the install.

I bought the SCPH-5552 console, installed and checked everything, when you turn it on, xstation still cannot start (((just like on the previous SCPH-5502...

ramapcsx2, 2022-02-15:
@Sklzz This modification requires good soldering. There should not be any experimentation such as what you show in your videos.
Consider having the installation done by a professional.
I can't help you, other than saying that the problem is your work.

Sklzz, 2022-02-15:
> @Sklzz This modification requires good soldering. There should not be any experimentation such as what you show in your videos. Consider having the installation done by a professional. I can't help you, other than saying that the problem is your work.

ok, I'm ready to send it to any specialist, but where can I find it?  I myself am in the Republic of Belarus.

Sklzz, 2022-02-15:
> @Sklzz This modification requires good soldering. There should not be any experimentation such as what you show in your videos. Consider having the installation done by a professional. I can't help you, other than saying that the problem is your work.

but I'm pretty sure it's not soldering.  Since this problem is on two consoles... I repeat once again, I checked everything with a multimeter, everything is fine, the problem is in the device...

ramapcsx2, 2022-02-15:
I can assure you that it's not a manufacturing problem. But that won't convince you, would it? :)
If shipping to Germany is doable, I can fix it.
I don't really know what to do otherwise.

Sklzz, 2022-02-15:
Where can I write to you about shipping?

Sklzz, 2022-02-15:
> > I can assure you that it's not a manufacturing problem. But that won't convince you, would it? :) If shipping to Germany is doable, I can fix it. I don't really know what to do otherwise.
> 
> I agree to send you two ps1 motherboards, the main thing is to make xstation work...

Where can I write to you about shipping?

Sklzz, 2022-02-15:
> I can assure you that it's not a manufacturing problem. But that won't convince you, would it? :) If shipping to Germany is doable, I can fix it. I don't really know what to do otherwise.

Where can I write to you about shipping?

ramapcsx2, 2022-02-15:
Okay, I'm currently down with a bit of the old Covid, but hop on to Discord and PM me there. I'll respond when I can :)
https://discord.gg/cmY3bMrS

Sklzz, 2022-02-16:
texted you on discord

ramapcsx2, 2022-03-05:
Update for myself, to keep track:
The xStation arrived and I'll install it on a spare PU-18. Then ship it back

ramapcsx2, 2022-03-11:
Oki, xStation installed and shipped back :)

closed 2022-03-11 by ramapcsx2