x-station/xstation-issues #195: Black Screen on

aacob2, 2022-02-11:
Installed xstation on scph-5502 and having this issue where we get no power to the xstation board and the console displays only black. We tested after lifting the pins on IC701 and the console worked perfectly and went to the playstation os. Once we got to installing the QSB and the wire we were unable to get past a black screen. I have tried both with and without the Xstation board installed and the issue remains. We have also tried removing the QSB and installing it again and the issue remained. 
Could the wire we used be incorrect? Or does it have to do with the resistor?
When you turn on the console on the xstation board
[image]
[image]
[image]

Also had a modchip installed previously which we removed
[image]
.

retrofank, 2022-02-11:
Looks like a QSB install issue tbh, if you say it booted to the MC/CD Player Screen without it installed and then it black screens upon installing it afterwards.  

The APLL wire is the clock signal not power.  The VCC point is where you should check.  But looking at the QSB pics (104 area) looks like it's bridged on two points and also, too much solder used on other areas.  Check continuity, as most likely a poor joint(s) or shorts may be present.

ramapcsx2, 2022-02-11:
I spot several shorts on the QSB, as well as some not connected points. 
Please either have it checked by someone with a little more soldering experience.

aacob2, 2022-02-12:
> Looks like a QSB install issue tbh, if you say it booted to the MC/CD Player Screen without it installed and then it black screens upon installing it afterwards.
> 
> The APLL wire is the clock signal not power. The VCC point is where you should check for power. But looking at the QSB pics (104 area) looks like it's bridged on two points and also, too much solder used on other areas. Check continuity, as most likely a poor joint(s) or shorts may be present.

Oh thank you. I'll see if we can fix the QSB issues but glad to at least hear it wasn't to do with the wire.

aacob2, 2022-02-12:
> I spot several shorts on the QSB, as well as some not connected points. Please either have it checked by someone with a little more soldering experience.

I will see if there is someone i can take it to for help. I live in Australia so I don't think there are much options I'm afraid.