x-station/xstation-issues #196: Fresh install, worked flawlessly exactly once, now only goes to mem card screen

3nemy11, 2022-02-13:
I feel bad instantly posting a new issue for help but I have been searching and trying everything I can find to try to fix this issue I'm having.

Today I installed the Xstation Rev 1.2a into a LATE PU-8 (23) SCPH-1001 with a Laserbear SD card extension kit. Everything seemed to go very smoothly. After buttoning everything back up and installing the newly formatted SD card with a single game I booted it up as the first test. The xstation booted up to the main screen and I refreshed game list to find Einhander. Selected Full Load and the game launched cleanly.

After playing around for a bit I decided to move some more game files over to the card. Moved about 4 then put the SD card into the Laserbear 3d printed slot. PS1 logo and then eventually a frozen memory card screen.

Took the sd card out, reseated it, same problem. Tried reformatting the card, same thing. Tried a different card, same issue.

I found a post with all the test points so I multimetered them all to the ribbon cable and all 1 through 20 (except for the 1 unused) showing continuity. Reassembled and retested, still the same problem.

I'm noticing the green LEDs on the Xstation blink dim and quick twice. And everytime I attempt a reformat disk with new loader.bin and update.bin I notice the Xstation seems to be writing a blank config.txt file.

I can supply any pictures to help! Thanks!!

birdybro, 2022-02-13:
Have you tested it without the laserbear extension kit?

3nemy11, 2022-02-13:
Hi Birdybro,
Thanks for the idea, I did give that an attempt but it still didn't work.

I'm sorry for the false alarm everyone. I managed to fix it by reflowing all the solder joints on the QSB. I've seen that work out for many people on here but I assumed since I tested continuity between all points they were solid. NOPE lol.

Anyways I'm glad these threads exist, they helped a ton.

closed 2022-02-13 by 3nemy11

Sklzz, 2022-02-14:
> Привет, Birdybro, спасибо за идею, я пытался, но все равно не получилось.
> 
> Прошу прощения за ложную тревогу у всех. Мне удалось это исправить, оплавив все места пайки на QSB. Я видел, как это работает у многих людей здесь, но я предположил, что, поскольку я проверил непрерывность между всеми точками, они были надежными. НЕТ, лол.
> 
> В любом случае, я рад, что эти темы существуют, они очень помогли.

> I feel bad instantly posting a new issue for help but I have been searching and trying everything I can find to try to fix this issue I'm having.
>
> Today I installed the Xstation Rev 1.2a into a LATE [...]

Wait, you said that before that you checked everything, and the connections were without breaks ...

3nemy11, 2022-02-15:
That's what surprised me too. I tested every point according to that test guide not only point to point but also through the ribbon cable and showed continuity every time. Reflowing those points somehow fixed the issue immediately though.