x-station/xstation-issues #198: Feature Request: .CHD support

CurkyHangles, 2022-02-28:
Hello Rama,

The xstation is a wonderful piece of hardware!  It made me dig my old 1001 out of storage and breathe some life back into the ole' girl.  Thank you for your amazing contributions to the retro-gaming and emulation communities!

I was curious if there would be a possibility of the xstation's software being updated in order to handle .chd files and if the hardware is powerful enough to do so, as there is probably some on-the-fly decompression that needs to happen.

Thanks again!

birdybro, 2022-02-28:
https://github.com/x-station/xstation-issues/issues/50 - Already requested :)

It does not seem to be feasible as the ESP32 is very limited in it's performance to be able to handle something like this while also doing everything else it has to do to emulate the optical disk drive.

CurkyHangles, 2022-02-28:
Thank you for the heads up!  I missed that issue.  I'll close this.

closed 2022-02-28 by CurkyHangles